vlachoudis/bCNC #217: Change directory

TilTisback, 2016-01-29:
Hello,
i found another bug in the file system.

when i try to open a folder bcnc shows me this opening window: 
[image]
now i click on the CAD button in the upper path to change that folder to D:\CAD

but instead of changing to the directory it shows me this error:
[image]

when i click directly on the D: button it changes the directory to this:
[image]

my workaround for this is a bit awkward:
i go to the .bcnc file in the users folder and change the dir = D:\CAD\Laserprojektor\CNC to dir = D:\CAD and restart bcnc
now i can select the subfolders correctly and open my new cnc job

i think the problem with this is as follows: the bcnc program is located on the E: drive while the cnc files are on the D: drive.

vlachoudis, 2016-01-30:
Thanks for the bug report. I will change to windows to see what I can do.

vlachoudis, 2016-01-31:
I have to find out how to scan for the windows drives in python.
For the moment you can change the drive by typing the drive name
C: or D:
in the field of "Filelname" and press Enter

vlachoudis, 2016-01-31:
I've changed the FileDialog to respect the drive letter of windows. Can you check if it works.
Still if you want to change drive you have to give it in the Filename

closed 2016-02-03 by vlachoudis