vlachoudis/bCNC #220: How to pocket the difference of two shapes

cclein, 2016-01-31:
I'm wracking my brain trying to figure out how to do the following:

Say I have two circles or squares, one inside the other. Is there a way for me to say pocket the inside shape and then do the difference between the outside shape and the inside shape to a different height? Specifically lower.

It was quite easy to do in JScut and I'm wondering if I'm missing something in bCNC.

vlachoudis, 2016-01-31:
No you are not missing anything. For the moment is not possible to do. Is in my TODO list :)

cclein, 2016-01-31:
OK Good. I'm really loving your software and the rapid development!

I'm going to try and create the gcode on jscut.org. Do you have any
tips on importing that into bCNC? I tried a long time ago and had
issues.

Cheers,

Chris

vlachoudis, 2016-02-01:
It importing doesn't work (strange) please send us the gcode so I could debug the problem.

cclein, 2016-02-01:
Actually importing worked much better than the last time I did it.
Unfortunately none of the blocks etc for each operation came over but I
could live with that. Bad news the code was too big for my Raspberry Pi :(

vlachoudis, 2016-02-01:
@cclein could you send me the input to check where it failed?

cclein, 2016-02-01:
It didn't fail. Just too large for the RaspberryPi. Became unresponsive and
slowed cutting down. The file is 10mb. I can't send it via github. Anywhere
else I could send it?

vlachoudis, 2016-02-01:
Can you zip it and send it via email. My address in the about of the program. There are a few things I want to optimize

vlachoudis, 2018-04-16:
@motanescu interesting idea.

Harvie, 2018-05-06:
This would truly be usefull feature to have.

@motanescu I tried to make circular pocket with nested circular island like this:

[image]

But the pocket failed to generate

```
Traceback (most recent call last):
  File "/opt/bcnc-git/Utils.py", line 413, in __call__
    return self.func(*args)
  File "/opt/bcnc-git/ToolsPage.py", line 1490, in execute
    self.tools.getActive().execute(self.app)
  File "/opt/bcnc-git/ToolsPage.py", line 769, in execute
    app.pocket(name)
  File "/opt/bcnc-git/bCNC.py", line 1758, in pocket
    msg = self.gcode.pocket(blocks, diameter, stepover, name)
  File "/opt/bcnc-git/CNC.py", line 3637, in pocket
    newpath.extend(self._pocket(path, -D*diameter, stepover, 0))
TypeError: 'NoneType' object is not iterable
```

motanescu, 2018-05-06:
Here is the code from my example. All things are grouped together in one layer. Maybe that matters

(Block-name: Header)
(Block-expand: 0)
(Block-enable: 1)
M3 S12000
G4 P3
(Block-name: Layer0)
(Block-expand: 0)
(Block-enable: 1)
g0 x0 y0
g1 z0 f600
g1 x-32.75 y0 f500
g1 x-32.75 y4
g1 x-21.25 y4
g1 x-21.25 y11.25
g1 x-21.277688 y11.813599
g1 x-21.360485 y12.37177
g1 x-21.497593 y12.919137
g1 x-21.687693 y13.45043
g1 x-21.928953 y13.960532
g1 x-22.21905 y14.444529
g1 x-22.55519 y14.897762
g1 x-22.934136 y15.315864
g1 x-23.352239 y15.694811
g1 x-23.805471 y16.030951
g1 x-24.289469 y16.321048
g1 x-24.799571 y16.562308
g1 x-25.330863 y16.752407
g1 x-25.878231 y16.889516
g1 x-26.436402 y16.972313
g1 x-27 y17
g1 x-27.563599 y16.972313
g1 x-28.12177 y16.889516
g1 x-28.669137 y16.752407
g1 x-29.20043 y16.562308
g1 x-29.710531 y16.321048
g1 x-30.194529 y16.030951
g1 x-30.647762 y15.694811
g1 x-31.065864 y15.315864
g1 x-31.44481 y14.897762
g1 x-31.780951 y14.444529
g1 x-32.071048 y13.960532
g1 x-32.312308 y13.45043
g1 x-32.502407 y12.919137
g1 x-32.639516 y12.37177
g1 x-32.722312 y11.813599
g1 x-32.75 y11.25
g1 x-32.75 y4
g1 x-32.75 y0
g1 x-40 y0
g1 x-40 y20
g1 x-18.650137 y20
g1 x-18.653927 y8.975452
g1 x-18.650137 y8.871082
g1 x-18.725924 y8.490086
g1 x-18.745615 y4
g1 x-18.669828 y3.618995
g1 x-18.577824 y3.446867
g1 x-18.454006 y3.295995
g1 x-18.303134 y3.172177
g1 x-18.131005 y3.080173
g1 x-17.75 y3
g1 x-9.754386 y2.999991
g1 x-9.371702 y3.076111
g1 x-9.198815 y3.168521
g1 x-9.047279 y3.292884
g1 x-8.922916 y3.444421
g1 x-8.830506 y3.617307
g1 x-8.754386 y3.999991
g1 x-8.774077 y8.490086
g1 x-8.846074 y8.975452
g1 x-8.965299 y9.451424
g1 x-9.130603 y9.913418
g1 x-9.340394 y10.356984
g1 x-9.884948 y11.171967
g1 x-10.214466 y11.535534
g1 x-10.578034 y11.865053
g1 x-10.972149 y12.157349
g1 x-11.393017 y12.409607
g1 x-11.836583 y12.619398
g1 x-12.298577 y12.784702
g1 x-12.774549 y12.903927
g1 x-13.259915 y12.975924
g1 x-13.75 y13
g1 x-14.725452 y12.903927
g1 x-15.201424 y12.784702
g1 x-15.663417 y12.619398
g1 x-16.106984 y12.409607
g1 x-16.527851 y12.157349
g1 x-16.921967 y11.865053
g1 x-17.285534 y11.535534
g1 x-17.615053 y11.171967
g1 x-17.907348 y10.777852
g1 x-18.369398 y9.913418
g1 x-18.534702 y9.451424
g1 x-18.653927 y8.975452
g1 x-18.650137 y20
g1 x0 y20
g1 x0 y0
g0 z2
(Block-name: Footer)
(Block-expand: 0)
(Block-enable: 1)
M5
G0 Z20

Harvie, 2018-06-25:
Wow, this is real PITA. At first it didn't really affected me, but with every new project this gets more and more annoying...

Harvie, 2018-07-02:
Just found the f-engrave is also written in python, however i am not sure if it would be possible to reuse some of it's code... Probably not :-(

Harvie, 2018-07-03:
What about adding button that will mark any path as "island":

[image]

And then having tool which will split any selected segment at intersection with island path and make it move to safe Z and back down (up on odd crosses, down on even crosses), maybe even reusing the code for cuting tabs. Once done you can append the island path to the end of the code, so island surface get finished.

Harvie, 2018-07-03:
Even better algorithm:

[image]

Harvie, 2018-07-03:
good news lib/bpath.py already has these :-)
intersectSelf(self)  # intersect path with self and mark all intersections
intersect(self, other) # Intersect with another segment, returns two points
isInside(self, P) #Return true if point P(x,y) is inside the path

Harvie, 2018-07-04:
alpha version of shape intersection plugin (technology preview for pockets with islands):

[image]

Harvie, 2018-07-04:
I obviously needed some sleep. I woke up, fixed single loop and boolean operation on path now works predictably! If i am correct, the pocket with islands is very close :-)

Harvie, 2018-07-04:
Just added way to tag path as an island (also sets color to red) for future purposes:

[image]

Harvie, 2018-07-04:
Tabs are now implemented as Path, so we can have any shape as soon as there will be code to asign the path to the right variable...

For now it has some caveats, which can be fixed in future:
  - tab can't be modified as the new tab format is not human readable (this can be fixed as soon as everything stabilizes)
  - tabs don't move with path (this can be fixed as soon as we add Path.move() method to lib/bpath.py)
  - CNCCanvas currently renders only bounding box of the tab, but it has no effect on tab functionality

Harvie, 2018-07-04:
I've implemented the feature with custom tabs for now. You can even get basic island if you apply such tab on pocket!!!!

[image]

Harvie, 2018-07-05:
Aha! Much smoother now:

[image]

Harvie, 2018-07-05:
It's funny how it handles self intersecting shapes as the star here. i'd call it a feature:

[image]

Harvie, 2018-07-05:
I think this pretty much solves this ticket... however i want to continue by creating more effective toolpath as i demnonstrated earlier...

Harvie, 2018-07-08:
The new optimized island pocketing would need the offset math to work flawlessly, so it's currently blocked by #903

Harvie, 2018-07-08:
[image]

I simplified the use of islands. now you don't need to create tab object from islands...

TGPSKI, 2018-08-11:
Thanks for this PR @Harvie - made quick work of the SF Giants logo w/ island.

[image]

Harvie, 2018-08-11:
@TGPSKI You're welcome! I haven't tried it on real machine yet. Nice to see it's already usefull. Do you have some thoughts on the process? I am bit concerned about the fact that it currently does lots of plunging. The edges look bit chewed, that's what i was afraid of... Have you noticed something strange or unwanted happening during this milling strategy?

Harvie, 2018-08-16:
@TGPSKI maybe try to use outside profile to make the island slightly bigger, so there's still some material left for final pass, which you can do using smaller profile of the island. That way you'll get better surface finish on island.

MARIOBASZ, 2018-08-20:
It's good to see it in a real job, I have not used it yet. From what you say, maybe I'm writing something silly that is already instrumented, in the islands is compensated the diameter of the tool, right?
I was rereading above, in your first steps, at the intersection of routes, I continued skirting the island instead of raising Z, can you make a version that follows that type of route?

Harvie, 2018-08-20:
> the islands is compensated the diameter of the tool, right?

nope. you can apply profile operation before making island

> you continued skirting the island instad of raising Z

i am working on it. see https://github.com/vlachoudis/bCNC/issues/220#issuecomment-402129911

MARIOBASZ, 2018-09-07:
Excellent! I congratulate you! great job

Harvie, 2018-09-07:
Now i've fixed difference plugin for cases, where island splits shape in two:

[image]

However real island aware pockets are still not possible. I have to rewrite the pocket code completely.

MARIOBASZ, 2018-09-07:
what does he do, which is wrong for you? surrounds the island that cuts the figure? If so, I think it is a useful utility or variant, do not discard it!

MARIOBASZ, 2018-09-14:
Will it also have the capacity to surround the island in the exterior profile operation?

MARIOBASZ, 2018-11-16:
Uau !, I have advanced with operations of pockets and islands!
Respect un-linearized circles.
Problem: If in a block, the pocket is separated into two sectors (figure on the right) a wrong cut (diagonal here) is generated. Solution that I imagine is correct: that the pocket operation generates separate blocks for each sector.
Curiosity: If I select the right one plus the one external, the pocket operation has been successfully sequenced. But I do not know if this is random.

[image]

Harvie, 2018-11-16:
@MARIOBASZ cool. I know how to improve:

1.) Do inside profile of shape
2.) Check if they collide and make difference if needed
3.) Do outside profile of island
4.) Check if they collide and make difference if needed
5.) GOTO 1.)

That will give you nicer outline of island. Like here: https://github.com/vlachoudis/bCNC/issues/220#issuecomment-402129911

leventelist, 2020-06-15:
Is this feature implemented in current master?

Harvie, 2020-06-16:
Just the custom shaped tabs. Boolean operations are not ready yet. TBH i didn't played with the code for like two years... Maybe it is time to continue, but i am very busy right now :-/

leventelist, 2020-06-16:
So I want to mill a text, and letter O and P is not possible then? Do you know any workaround to achieve that? But thanks anyways for the good work!

Harvie, 2020-06-16:
You can, eg.: https://github.com/vlachoudis/bCNC/issues/220#issuecomment-412243226

but the toolpaths are not optimal = will take bit longer and leave bit worse surface finish

MARIOBASZ, 2020-06-16:
As @Harvie told you, her work on the island is a solution. Does the difference not work? I think if (plugin development itself) ... or I'm using it with some slight modifications (difference, solid join, intersection), I haven't uploaded what I did, the reasons:
1- unresolved error if the silhouettes share any segment
2- reduced knowledge of python to solve it, many times over time I keep learning to solve problems, I still have little time
3- my git branch was outdated, I tried to load a plugin, I couldn't, I didn't try again
I can copy the plugin here, but see if either of the two alternatives works.

#1175

leventelist, 2020-06-16:
Okay. Can you help me how to achieve this then? So I have my DXF file
with the shape. I then toggle the inner ring to be an island, and then
I execute a pocket operation. I experience that the whole outer shape
is removed. What do I miss?

Harvie, 2020-06-16:
Then you have to create another cutting operation just for the inner shape... THis is not automatic yet...

Harvie, 2020-06-29:
Note to self: https://ideolalia.com/2018/08/28/artifex.html

Harvie, 2021-02-23:
This was implemented

[image]

Huge thanks to @DodoLaSaumure !

@TGPSKI Perhaps you can try to cut that SF Giants logo again with this great new tool to see if the results are more crisp now :-)

closed 2021-02-23 by Harvie

Harvie, 2021-02-23:
Also "profile" and "pocket" tools will be removed in future releases, because "offset" tool have replaced them.

MARIOBASZ, 2021-02-23:
Pierre, have you been able to solve the sectors that were spaced greater than the correct one?

DodoLaSaumure, 2021-02-24:
> Pierre, have you been able to solve the sectors that were spaced greater than the correct one?

Sorry I do not understand the question. 
Do you mean this ? => Yes. 
[image]

Pierre

TGPSKI, 2023-08-09:
[image]

Any thoughts on this output from the offset profile operation? (using islands / toolpath circles over island in a weird way.

Going to try to build rar8000's latest and give it a spin on the CNC if I can get these paths figured out

TGPSKI, 2023-08-09:
And @Harvie I re-built my CNC machine in the past two months, happy to do some test cuts on 12" x 12" stock