vlachoudis/bCNC #221: Probe dialog box

Ozzy62, 2016-01-31:
This may just me having an issue, bit would it be possible to ad a dialogue box asking if you remembered to connect the wire to the toolbit. I just, for the countless time, missed it again.
Or maybe just point me in a direction on how I can ad it myself.

Regards
Svend
PS: Thanks a great piece of software, have just switched from Mach3, I a happy dude.

vlachoudis, 2016-01-31:
Of course it is possible, I am simply wondering such messages are helpful or simply annoy the user.
If you want to give it a try, you can add a tkMessageBox.showinfo() call in the ProbePage.py
class ProbeFrame, method "def probe()"

Nate-Bro, 2016-02-01:
The other day I was trying to do a pcb, and broke 5 bits!  I'm too forgetfull when it comes to that dang wire....

It might be helpful to display it once then have a "never show again" option people can check off.

https://github.com/vlachoudis/bCNC/issues/221#issuecomment-177599724

vlachoudis, 2016-02-01:
@Nate-Bro that can be an option. I will do.

Ozzy62, 2016-02-01:
I guess you are right vlachoudis, it may just annoy the most. But a solution as Nate-Bro comes up with is perfect.

vlachoudis, 2016-02-04:
@Ozzy62 in the new master I have a dialog that pop-up when you click on probe. And you can select if you want to dismiss it or keep it. For the moment is Yes/No which I don't like much, I should put a checkbutton for that.

Ozzy62, 2016-02-04:
I can only thank you (having plunged the bit into the PCB for, insert any stupidly large number, times). I will take a look as soon as possible. I have already milled a PCB with your software and it was perfect. Thank for your effort to bring software like this to us all.

closed 2016-02-11 by vlachoudis