vlachoudis/bCNC #228: Python Version?

ericrector, 2016-02-08:
First thanks for the time and effort with this project!

Which python version does this run under? I am using, Python 3.3.3 (v3.3.3:c3896275c0f6, Nov 18 2013, 21:19:30) [MSC v.1600 64 bit (AMD64)] on win32

When I run the scripts I get modules that are not found:
ImportError: No module named '__builtin__'

I changed:

import __builtin__
to
import Builtins

(http://docs.pythonsprints.com/python3_porting/py-porting.html#name-changes)

then I get:
Traceback (most recent call last):
  File "C:\xxxxxxxxxxxxxxxxx\bCNC.py", line 27, in <module>
    __builtin__._ = gettext.translation('bCNC', 'locale', fallback=True).ugettext
AttributeError: 'NullTranslations' object has no attribute 'ugettext'

also httplib is now urllib????

Do you have any advice?

Effer, 2016-02-08:
Any 2.7 version is working. Efforts are made to work with 3.3 but not high priority for now. Give a try and report any feedback.

ericrector, 2016-02-09:
I'll see how I can help.

vlachoudis, 2016-02-09:
@ericrector most of the functionality is already there with python3 (not fully tested yet). The main problem comes from the core sender routine, where the serial interface expects a "bytes" type in python3 and the strings are unicode, while in python2 expects a "str".
I didn't any good way of making it work for both versions with the same code, without making an extensive use of try: except:

voneiden, 2016-02-10:
@vlachoudis I'm not in the loop on the details of the code in question, but I'll just throw this here.

```
Python 3.5.1 (default, Feb  2 2016, 15:28:26)
>>> type("foo".encode("utf8"))
<class 'bytes'>

Python 2.7.5 (default, May 29 2013, 21:11:16)
>>> type("foo".encode("utf8"))
<type 'str'>
```

Encoding a python 3 normal unicode string (str) results in bytes. Encoding a python 2 regular string (str) results in str. Encoding a python 2 unicode string (unicode) results in str. I have a hunch that's the method you've been looking for.

vlachoudis, 2016-02-10:
Thanks @voneiden this could do the conversion. However there are a lot of checks in the code for "unicode" type that doesn't exist in python3 which raise NameError exceptions.
Maybe a dirty hack would be to create a builtin type unicode equal to str.
There are several places we have to place conditions for either python2 or python3.
All these will end up with a messed up code.
Personally I believe that the best option would be one day to move completely to python3 and stop supporting python2.

voneiden, 2016-02-11:
Alright. I'll try to find some time to sit down and get familiar with the codebase, maybe I can come up with something. I agree on moving towards python3. Legacy python2 support would be nice to keep, but it should have minimal impact on the main code.

vlachoudis, 2016-02-11:
I was a bit afraid of doing the jump right now. I wanted to get operational all the features I wanted the program to have, and then move to python3.

voneiden, 2016-02-11:
OK, I'll keep an eye on the situation.

ericrector, 2016-02-16:
Hi guys,

I didn't mean to open a can of worms.  I can try to help. I have a strong
back ground in c, cpp and java.  I am pretty new to python.  I don't think
it will be too difficult to learn, but finding time, as you all know, is an
issue. Maybe someone has a suggestion on how we can work together to help
create a Python 3 version.

But if Vasilis wishes are to wait, I can respect that.

vlachoudis, 2016-02-17:
@ericrector thanks for your offer to help. I just want to avoid having a code which is a patchwork in order to support both python versions while I am still some features are not fully implemented. I would like at some point to make the jump to python3 and stop the support of python2.

I would say the first step would be to modify all no-cricital parts to be python2 and python3 compatible. I am trying already to do this, but extra help is welcome. Then start working on the critical parts of the code and convert them only to python3 (especially the sender)

DasWookie, 2016-08-05:
Launch on python v2.6.6 is throwing an error with latest code:
$ ./bCNC.py
Traceback (most recent call last):
  File "./bCNC.py", line 2374, in <module>
    if sys.version_info.major != 2:
AttributeError: 'tuple' object has no attribute 'major'
$ python --version
Python 2.6.6

---------------------------------

On Python version 2.7.9 the 'major' tuple object resolves, so no problem, and bCNC runs without any issues with the version check code.

vlachoudis, 2016-08-06:
@DasWookie thanks for reporting. I've modified the code to ask the sys.version_info[0] instead of the .major. Now it should work also on 2.6

atrueresistance, 2017-09-01:
Is support for 3 working now?

vlachoudis, 2017-09-04:
Nope, a python 3 will require quite a lot of work and testing since the string now is unicode (2bytes), while on the controller it has to communicate with byte streams.

atrueresistance, 2018-03-30:
@vlachoudis  I've started a 100 days of code challenge and once I'm more familiar with the language in general I'll be looking into this. 

I'm wondering if setting `tosend` in Sender.py `def serialIO(self)` to a byte type would be enough?

[image]

I use bCNC almost daily, so should be able to test on another laptop of mine.

Harvie, 2018-11-22:
Just for my internal record: https://www.root.cz/clanky/domena-cz-slavi-25-let-zacalo-to-souborem-hosts-txt/

atrueresistance, 2019-01-09:
@Harvie I see you made a branch V3. Is this supposed to be the location to develop/test against? 

```
$ python bCNC.py
Traceback (most recent call last):
  File "bCNC.py", line 1, in <module>
    __main__.py
NameError: name '__main__' is not defined
```

Harvie, 2019-01-09:
@atrueresistance it's not really up to date. also we never got it to work properly. i've managed main window to show, but almost everything was broken.

Harvie, 2019-11-27:
Good news everyone! I think i've fixed most of the problems with bCNC on python3. Now i need you all to heavily test latest git master version on both python2 and python3, so we can be sure it's ready for release!

MARIOBASZ, 2019-11-27:
I'm glad, congratulations!

Harvie, 2019-11-27:
There seems to be bug with serial commmunication in python3, can someone tell if it also affects the python2 ? I have some problem with gcc, so i can't install grbl to test it right now.

onekk, 2019-11-27:
Ok the bug is in the coding of the line in Sender.py
`
# Anything to receive?
            if self.serial.inWaiting() or tosend is None:
                try:
                    line = str(self.serial.readline()).strip()
                    #print("Received line > ", line)
`

the commented print results in b'' value in python 3 (Actually Python 3.5.3 in Debian 9). more to come..

onekk, 2019-11-27:
`                try:
                    self.serial.write(tosend.encode())
                except AttributeError:
                    self.serial.write(tosend)`

This seem to solve the mixed problem, something is generated as str and something as byte, so if it is byte the encode() will fail with Attribute Error.

Harvie, 2019-11-27:
I think we will need some wrapper around serial.write(), so we don't have to put the try/catch statement everywhere.

onekk, 2019-11-27:
Yes, it will be maybe better to analyze the tosend type and act accordingly:

`

                print("serial to send > ", tosend)
                print("serial to send type> ", type(tosend))

`

if you add these two line prior to try: except part, you will se that some "lines" are passed as str() some others are passed as byte b'', how to figure out where this lines came from?

see this sequence from the command line:

`

serial to send >  $#

serial to send type>  <class 'str'>
serial to send >  $G

serial to send type>  <class 'str'>
serial to send >  $J=G91 X10.000000 Y10.000000 F100000

serial to send type>  <class 'str'>
serial to send >  b'$G\n'
serial to send type>  <class 'bytes'>
Controller state changed to: Jog (Running: False)
Traceback (most recent call last):
  File "OKKCNC/__main__.py", line 2813, in monitorSerial
    self._monitorSerial()
  File "OKKCNC/__main__.py", line 2641, in _monitorSerial
    line = line.rstrip("\n")
TypeError: a bytes-like object is required, not 'str'
Controller state changed to: Idle (Running: False)
serial to send >  $#

`

onekk, 2019-11-27:
For some code that deal with python differences, I have resolved setting a "global" var imported from a module in my case OCV.

the IS_PY3 is set in the first place when you test for python 2 or 3.

as example here you a part of code what deal with dictionaries, as in python2 > 2.7 viewitems is working as items() in python3 

as reported by this note:

The objects returned by dict.viewkeys(), dict.viewvalues() and dict.viewitems() are view objects. They provide a dynamic view on the dictionary’s entries, which means that when the dictionary changes, the view reflects these changes.

 `

        if OCV.IS_PY3:
            iterlist = OCV.PLANE.items()
        else:
            iterlist = OCV.PLANE.viewitems()

        for key, val in iterlist:
            bal_text += "{0} > {1}\n".format(key, val)

 `

maybe to cope with differents beahviour in delicate places a simila approach could be used.

Regards

Carlo D.

onekk, 2019-11-27:
This sendGcode class for Sender.py

seems to work for python3 and Python2

"""
    def sendGCode(self, cmd):
        """
        Send to controller a gcode or command
        WARNING: it has to be a single line!
        """
        if self.serial and not self.running:
            c_type = type(cmd)
            if OCV.DEBUG is True:
                print("Cmd      > ", cmd)
                print("Cmd type > ", c_type)

            if isinstance(cmd, tuple):
                self.queue.put(cmd)
            elif isinstance(cmd, str):
                self.queue.put(cmd + "\n")
            else:
                self.queue.put(cmd + b"\n")

"""

Harvie, 2019-11-27:
I've added `serial_write()` which is wrapper for `serial.write()`. So we can fix this in single place. But i can't get it working in python3. There seems to be catch-22 problem in pyserial: https://github.com/pyserial/pyserial/issues/467

Any ideas how to fix this?

Harvie, 2019-11-28:
OK guys. I've figured it out. Now serial connection should work well in both python2 and python3.
I am waiting for you to search for more issues (both py2 and py3) before i release this!

Problem is the `spy://` pyserial mode is broken in `?raw` mode. So until they fix it in https://github.com/pyserial/pyserial/issues/467 , you have to use it without `?raw` modifier on python3

onekk, 2019-11-28:
This is working at least for python3

```
   def serial_write(self, data):
        """Serial write"""
        #print("W "+str(type(data))+" : "+str(data))

        if OCV.IS_PY3 is True:
            c_type = type(data)

            if OCV.COM_DEBUG is True:
                print("serial to send > ", data)
                print("serial to send type> ", c_type)

            if isinstance(data, bytes):
                ret = self.serial.write(data)
            elif isinstance(data, str):
                ret = self.serial.write(data.encode("ascii"))
        else:
            ret = self.serial.write(str(data))

        return ret

```

The logic is inverted respect to bCNC

as the problems arise only in python3 with the different str() treatment, the routine check for the data type and act accordingly, encoding only if needed.

In some part bCNC send b"" strings that in python2 are treated as str() while in python3 are different data, but

- with this serial_write, maybe all the b"" occurencies in bCNC have to be reverted to normal str() and converted only if needed here?

- maybe even a serial_read would be written that permits to convert the data in only one place, as checking for str() and b"" is always false in python3?

Regards

Carlo D.

Harvie, 2019-11-28:
@onekk Don't worry. I've got it working. Also i don't think i need custom serial debuging code, because pyserial has the `spy://` feature...

Now it looks like this and i think it works nice:

```python
        def serial_write(self, data):
                #print("W "+str(type(data))+" : "+str(data))

                if sys.version_info[0] == 2:
                        ret = self.serial.write(str(data))
                elif isinstance(data, bytes):
                        ret = self.serial.write(data)
                else:
                        ret = self.serial.write(data.encode())

                return ret
```

Harvie, 2019-11-28:
> maybe all the b"" occurencies in bCNC have to be reverted to normal str()

@onekk i think `bytes` is preffered data type for serial communication. so don't really. if you specify the commands as bytes, you can be sure that they don't get messed up by encoding. Feel free to try latest git master. I think it works well.

onekk, 2019-11-28:
> @onekk i think `bytes` is preffered data type for serial communication. so don't really. if you specify the commands as bytes, you can be sure that they don't get messed up by encoding. Feel free to [...]

If you check with the print statements enabled, you will not that sometimes data are passed as str() and sometimes as b"" and this raise the problem, most of the time.

I've noted in the process of error checking that in python3 you could specify some parameters to the open() routine to do encodings there, maybe it is good to converted directly on input.

b"" data are not very friendly with the "".format() syntax, used in many places (bCNC uses the 'old' % syntax, but I presume it is working in the same manner).
so you can't write 
```
b"something {0} {2}".format(vara, varb)
```
and is difficult to process all the "string"  in bCNC as b"" data.

if you are sure that all data that reach serial_write are str() the python3 check for the type is useful.

Regards

Carlo D.

Harvie, 2019-12-02:
So i've been trying the python3 version on real machine and it seems to work, however it produced following message:

```
Traceback (most recent call last):
  File "/home/harvie/Temp/bCNC/bCNC/__main__.py", line 2501, in monitorSerial
    self._monitorSerial()
  File "/home/harvie/Temp/bCNC/bCNC/__main__.py", line 2350, in _monitorSerial
    line = line.rstrip("\n")
TypeError: a bytes-like object is required, not 'str'
```

so i will have to fix it

Harvie, 2019-12-02:
So i've just published python3 compatible version **0.9.14.300** at pypi:
https://pypi.org/project/bCNC/

Harvie, 2020-04-06:
Ok, python3 support seems to work, there might be some problems, but that is to be handled in separate issues, so i will close this one.

commit e7c929fe (2019-11-28): Python3 serial.write fix #228
closed 2020-04-06 by Harvie