vlachoudis/bCNC #231: tool change settings

ericrector, 2016-02-09:
Here is my code:

```
G00 Z25.4000 
M06 T01  ; 0.8128 
G01 Z0.0000 F508.00 
G00 Z2.5400 
M03
```
here are my tool change settings:

[image]

The tool change routine runs; it lowers the tool until it touches the touch pad (1.1mm thick) than returns to the change position.

It all seems to work up until the point the the CNC starts running again where it returns to previous z and not the new z that is probed for.

I've tried many different setting, but I just can't find the right setting.

Are there any instruction on how to set up the tool change?

Thanks for you help.

vlachoudis, 2016-02-09:
Did you do the following before starting the run:
1. "zero" the z axis touching the surface of your stock
2. run a "calibrate" cycle to find the Dz between your stock zero and touch pad

ericrector, 2016-02-09:
Thanks, I think that was the key.

 First I ran a small macro to find Z0:
```
G38.2 Z-10 F25
G10 P0 L20 Z1.1
G0 Z5
```
then I ran the "calibrate" as you said. 
- In that order is works.

Do you have a touch plate function?

Also do you have a help page, What is TLO and WCS, or the different "policies" for the tool change?

vlachoudis, 2016-02-09:
What do you mean by touch plate function? There is a "Probe" function that allows that does exactly what you did to find the Z0.

Unfortunately the wiki is a bit behind (soft way of saying). Developers are very bad in writing documentation :)
bCNC is under heavy development, with new features that are not documented at all.
e.g. the image you send is already old. In the current version you have also the "orientation" information in the probing tab.
But you are welcome to help and writing it directly your findings on the Wiki page.

TLO is tool length offset, WCS is work coordinate system (standard terminology in CNC)
The policies are how the bCNC is expanding or not the tool changes commands.
- `Send M6 commands` will send the line as is to the controller. Grbl will refuse it with in error.
- `Ignore M6 commands` will skip completely the M6 and will not send it to the controller. more or less what other senders are doing
- `Manual Tool Change (WCS)` bCNC will expand the M6 command to a tool change cycle, and in the end will re-adjust the WCS to match the new offset of the tool. It has advantages of being persistent between reset of the controller, however the WCS will always reflect the tool mounted on the spindle and no longer the workpiece.
- `Manual Tool Change (TLO)` bCNC will expand the M6 as before and instead of setting the WCS will set the TLO variable of Grbl to match the difference in offset between the tool you did the calibration and the new one. This is the standard way of working on big machines were the WCS reflects the piece mounted and not the tool. However you would need to have a calibrated "probe" tool that gives you always the same reference. The TLO will be lost on reset of the controller.

lalo-uy, 2016-02-09:
This thred ned to be edited some how and moved to the wiki.
Will try to do so, but my inglish is not perfect.

flaplante, 2016-02-09:
I added the information to the bottom of the Tool change page of the WIKI.

Fred LaPlante, WA1DLZ

ericrector, 2016-02-09:
Thanks a million!

closed 2016-02-09 by ericrector