vlachoudis/bCNC #232: pythonw issue

sjj47, 2016-02-10:
I've been using bCNC version 0.7.3 for a while now with no problems.  However, I just downloaded version 0.9.0 to pick up the Probe > Autolevel fixes, and found an issue.

When I tried to run bCNC.bat I got an error - "Windows can't find 'pythonw'."  I tried adding the Python27 directory path to the Windows Environmental Variables PATH.  Then I could start bCNC, but I got an Warning saying that pythonw was blocked by my security software.  bCNC seems to run OK without it, so I'm not sure what is going on.

In a quick search of the web I found references to viruses associated with pythonw.  I don't know Python, so this may just be my ignorance.  Is python needed for bCNC?  Is this a change since version 0.7.3?

vlachoudis, 2016-02-10:
I've recently changed the bat from using python to use the pytonw. bCNC is written in python and it is needed to run. python is the console executable of python and pythonw is running without console.
So I wanted to get rid of the useless console window, using pythonw since bCNC traps all errors internally. I don't understand why your antivirus was blocking pythonw?
Apparently, there are still a couple of references on the console inside bCNC so I just restore it to the previous value. Please re-download and should work.

Effer, 2016-02-10:
Hi @sjj47 

* Recently the bat had been changed to start python in a different way. Pythonw is like python but it doesn't show that black prompt window that looks so nasty. It may be that the PATH variable was not set correctly with your python installation.

* The security prompt could be raised first time because pythonw was not registered as trusted, or even because latest version implements a auto update feature that look at the web rising some warning with some strong antivirus program.

* Pythonw is not a virus as much it could be Java or .Net, it mostly depends from the code you run on it. 
bCNC need it but ,all the code (lot of code :)  is in clear and readable, 

so enjoy bCNC and mill a lot!

vlachoudis, 2016-02-10:
@Effer probably this was the cause of the "Stop" traceback your reported, since the console stream was missing.

Effer, 2016-02-10:
@vlachoudis I'll try to reproduce and let you know.

sjj47, 2016-02-10:
I re-downloaded the zip file and ran the new version.  No errors or warnings.   I am seeing the python.exe terminal window now behind the bCNC window.

Thanks for the quick response!  You guys are great!

Effer, 2016-02-10:
@vlachoudis indeed the pythonw hide the other "Stop" issue , and here the console ouput:

Traceback (most recent call last):
  File "C:\Users\CNC\Desktop\bCNC.git\bCNC.py", line 2198, in monitorSerial
    self._monitorSerial()
  File "C:\Users\CNC\Desktop\bCNC.git\bCNC.py", line 2105, in _monitorSerial
    self._terminalCount = int(self._terminal.index(END).split(".")[0])
AttributeError: Application instance has no attribute '_terminal'

It happens after a stop during a huge cnc sending. Not yet tried with latest version.

Effer, 2016-02-10:
Good news @vlachoudis, latest version seems stable even for the "Stop" issue.

vlachoudis, 2016-02-10:
@Effer thanks for the confirmation. I need to slowly remove all stdXX output statements.
The _terminal was corrected on the recent update.
Now the terminal only displays up to 1000 lines.

closed 2016-02-10 by vlachoudis