vlachoudis/bCNC #233: v0.9.0 do not open

Durga000, 2016-02-11:
Hello,

I am using v 0.8.3 without any issue but v0.9.0 do not open when launching bCNC.bat.

I am Using Python 2.7, any ideas on ?

Thanks for your soft

sjj47, 2016-02-11:
Are you working in Windows?   I had a similar issue with v0.9.0.  My fix was to add the path to Python (in my case c:\Python27) to the Windows Environmental Variable PATH.   Then bCNC could find the needed Python files to run.  Don't know why v0.9.0 needed the path in PATH, but v0.8.3 did not.

vlachoudis, 2016-02-11:
Could you please redownload. It could be the change from python to pythonw as executable which was referred very quickly

Durga000, 2016-02-11:
I have re-downloaded but does not change still open and closed in the same time.

I also have add PYTHONPATH in the Windows Environmental Variable PATH ( C:\Python27;C:\Python27\Lib;C:\Python27\DLLs;C:\Python27\Lib\lib-tk) but without effect to my issue: 0.8.3 can open but not 0.9.0

vlachoudis, 2016-02-11:
Could you open a terminal "cmd" and launch manually the program. Most probably something is crashing at startup before the main window opens.

Durga000, 2016-02-11:
Here is the error message :

> 
C:\Python27>python C:\Users\Pa\Documents\CNC\bCNC-master-3\bCNC-master\bCNC.py
Traceback (most recent call last):
  File "C:\Users\Pa\Documents\CNC\bCNC-master-3\bCNC-master\bCNC.py", line 2324, in <module>
    application = Application(tk)
  File "C:\Users\Pa\Documents\CNC\bCNC-master-3\bCNC-master\bCNC.py", line 106, in __init__
    self.iconbitmap("bCNC.ico")
  File "C:\Python27\Lib\lib-tk\Tkinter.py", line 1693, in wm_iconbitmap
    return self.tk.call('wm', 'iconbitmap', self._w, bitmap)
_tkinter.TclError: bitmap "bCNC.ico" not defined
C:\Python27

Thanks,

vlachoudis, 2016-02-11:
It seems that for some reason cannot access the bCNC.ico. It should be on the main folder of bCNC.

Durga000, 2016-02-11:
the bCNC.ico file is well on the main folder...

Have try to re-install python but same issue, working with v.8.3 but not with v0.9.0

vlachoudis, 2016-02-11:
@Durga000 I cannot reproduce or understand what is the problem, why to work the 0.8.3 and not the 0.9. I don't know well windows, but I've just downloaded the latest bCNC and the newest python 2.7.11 expanded in bCNC in a folder click on the bCNC.bat and everything worked ok.
Are you sure is not permissions problems. Could you try to reset the permissions with
icacls * /t /reset

Durga000, 2016-02-11:
I understand how difficult it can be for you, no problem if you do not find the solution, you are already doing a great job with bCNC.

I have test .icacls but do not change anything.

Durga000, 2016-02-11:
I have downloaded another time v 0.9.0, delete the previous one, and expand the new one in a new folder, and guess what ?
It's works. 

Thanks for your support and for your dev on bCNC.

vlachoudis, 2016-02-11:
Mysteries of computers :) I believe that something was wrong in the file permissions.

closed 2016-02-11 by vlachoudis

Vladivarius, 2018-08-01:
I have found the solution for this, use absolute paths. 
Instead of self.tk.iconbitmap("bCNC.ico"), you should use self.tk.iconbitmap("C:\\bCNC\\bCNC.ico").