vlachoudis/bCNC #234: Tool change on latest master branch

atrueresistance, 2016-02-11:
Just pulled the latest master branch before shooting this video. https://youtu.be/qf4ZVV74A8s

What seemed to change? Before it was calibrate, and run for manual tool change. Pause never worked, but it was something I knew and expected (just remove that alligator clip quick). Changed to ignore M6 and back to cutting, but dang what happened? 

Let me know if you want the file.

vlachoudis, 2016-02-11:
@atrueresistance normally nothing happened in the Tool Change. I don't understand why for you the pause is not working. Do you use the english version? I saw one issue when unicode characters are used.

vlachoudis, 2016-02-11:
@atrueresistance I've tried to do a quick fix, but I can only test later in the evening. Could you test it and let me know.

Just another remark, you don't need to remove the M9 and M30 commands manually you can
add in your .bCNC config file
in the [error] section the lines
M9 = 1
M30 = 1
0 = STOP, 1 = SKIP

atrueresistance, 2016-02-11:
@vlachoudis Thanks, I'll pull down the master this morning and test it out and report back. Thanks for the error file info.

atrueresistance, 2016-02-11:
@vlachoudis Manual toolchange is working again. Pause still isn't working, this is the correct setting right? 
[image]

vlachoudis, 2016-02-11:
Indeed this is the correct settings.
It should pause twice once before and once after the tool change, by sending an "!" to grbl.
There was a discussion #163 similar, that by replacing the %pause in CNC.py internal bCNC command to "m0" pause was working correctly.

vlachoudis, 2016-02-11:
@atrueresistance  I've just test it on my machine and it is working correctly. However
I've replaced the feed hold with m0 as it was suggested by @mano1979, and modified the code to correctly show the messages. Could you check if it works for you now?

vlachoudis, 2016-02-12:
I couldn't test this behavior on my machine, so for me it was working before and now, @atrueresistance could you confirm that is working for you so I can close the issue?

atrueresistance, 2016-02-12:
I didn't get a chance last night to try changing m0. Give me some time this weekend.

mano1979, 2016-02-12:
I finally had the time to test it. After changing the "!" command to "m0\n", i get a pause before probing and after probing. I have set it to "only before probing" in bcnc. It really isn't a problem because we want time to remove the alligator clip, but i just wanted to mention it.

Also it would be great to have the message bar tell you the tool you have to change to. Now it quickly flashes a m06 message but before you can read it, it is gone.

vlachoudis, 2016-02-13:
Are you running the latest grbl? For me it shows the message until you click resume.

mano1979, 2016-02-14:
I don't believe so. Since i use "grbl-for-cyclone" ( arduino mega2560 and ramps1.4 )

It says version 0.9 though.

atrueresistance, 2016-02-19:
Sorry for the late response. Just pulled the latest master. The machine does in fact pause now after hitting the touch plate, but the play button is grayed out. I am running grbl .9g. @mano1979 where did you change that command?

atrueresistance, 2016-02-19:
[image]
[image]

Found it in CNC.py. It pauses, but just clocks. Play or terminal is not accessible.

mano1979, 2016-02-19:
Indeed cnc.py is the file.

To resume/un-pause you right-click on the message bar where it says queue. Then you can click resume.

It would indeed be handy-er to have the pause button do this....

vlachoudis, 2016-02-19:
Or even if you right click on the status message you can access the pause/resume buttons

mano1979, 2016-02-19:
Indeed. That is what i meant.

atrueresistance, 2016-02-19:
Super useful. did not know that was there

closed 2016-02-19 by atrueresistance