vlachoudis/bCNC #236: UnicodeDecodeError v0.9.0

Durga000, 2016-02-11:
Again...

I can now open v0.9.0, homing position, but the reported position are not correct and I cannot move motors.
I have tried to change range of movement ( we never know) and the changes cannot be saved.
Each time i re-open bCNC, the range values are the original ones not the modified ones.

I Don't know if the 2 problems are linked or not, (maybe is it again my laptop) hereunder the error message:

Traceback (most recent call last):
  File "C:\Users\Pa\Documents\CNC\bCNC-master-3\bCNC-master\Utils.py", line 343, in __call__
    return self.func(*args)
  File "C:\Users\Pa\Documents\CNC\bCNC-master-3\bCNC-master\ControlPage.py", line 303, in setX0
    self._wcsSet("0",None,None)
  File "C:\Users\Pa\Documents\CNC\bCNC-master-3\bCNC-master\ControlPage.py", line 350, in _wcsSet
    data=(_("Set workspace %s to X%s Y%s Z%s")%(WCS[p],str(x),str(y),str(z))).encode("utf-8"))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 1: ordinal not in range(128)

Thanks

vlachoudis, 2016-02-11:
I have the same problem on linux if I change the language. Once we started the internalization of the code I had to put explicitly the encode("utf8") function to pass the messages as utf8 to be displayed, otherwise tk was complaining. Now I have to remove the encode() to make it work. I don't understand what changed in between.

vlachoudis, 2016-02-11:
The encoding error was preventing from setting the value in the WCS. Now should be ok.

Durga000, 2016-02-12:
Sorry, I still have the issue. There is no more error indicated on the Python windows but the Wpos are wrong and I cannot move the motors...

Durga000, 2016-02-12:
Sorry, the issue is not on bCNC as I have the same trouble with v 0.8.3

Just need to find out what I have changed...

Durga000, 2016-02-12:
One of my axes was not well plugged... just feel stupid :-)

Once again Thanks for your work.

closed 2016-02-12 by Durga000

vlachoudis, 2016-02-12:
@Durga000  don't worry, thanks for anyway for the Unicode bug :)