vlachoudis/bCNC #241: bCNC leadin G3 instead of G2

eduardsik, 2016-02-17:
Hello, I found a bug. When generating G code from fusion 360 and leadin is enabled with arc (G18 G3 ...) draws G2 arc instead of G3

Here is the code, screenshots:
*******
Code from fusion 360 with Shapeok2 post processor:

 %
(gcode created by Fusion 360)
(Post processor file version: 1.31)
(Program name: 1.nc)
 
(----- Machine Info-----)
(vendor: Inventables)
(model: Shapeoko 2)
(desc: descktop CNC router)
 
( ------ Tool Table ------)
(T12  Dia=12 CR=0 - ZMIN=-1 - flat end mill)
 
G90 G94
G17
(Metric Units - mm)
G21
G54
G0 Z15

(Face1)
M9
(T12 M06 - not supported by Shapeoko/GRBL)
S5000 M3
M8
G0 X69.6 Y4.128
Z15
Z5
G1 Z2 F1000
G18 G3 X66.6 Z-1 I-3
G1 X60
X0
G17 G2 Y14.143 J5.007
G1 X60
G3 Y24.157 J5.007
G1 X0
G2 Y34.172 J5.007
G1 X60
G3 Y44.186 J5.007
G1 X0
G18 G3 X-3 Z2 K3
G0 Z15
G17
M9
%

******

bCNC screenshot

[image]

OpenSCAM screenshot with same code:

[image]

If I change manually G3 to G2, the bCNC acts as original code.

vlachoudis, 2016-02-17:
Thanks @eduardsik I've uploaded a fix for that.

eduardsik, 2016-02-18:
Thanks. What is the proper way to update from git to raspberry pi? Pull down in the same dir?

vlachoudis, 2016-02-18:
Yes, just git a "git pull" on the directory

closed 2016-02-18 by vlachoudis