vlachoudis/bCNC #242: bluetooth connection

uli99, 2016-02-17:
Hi,
I'm using bCNC for a while now and I must say it is amazing!

Now to the problem I have. I'm trying to use a bluetooth module (BTM-222) to connect with my GRBL-Controller. I'm working under linux and the module is paired and I can talk via the bluetooth module to a controller with a serial program (moserial).

The serial port appears as '/dev/rfcomm0' in the system.

The problem is this port doesn't show up in bCNC. I tried adding  '/dev/rfcomm' to the comports() function (in Utils.py) but it seems that it is using serial.tools.list_ports and that doesn't show/ consider '/dev/rfcomm'.

What can I do to get '/dev/rfcomm0' to show up?

I'm using Linux Mint 17.3 and bCNC 0.9.1

Thanks again for this great app!

Uli

Effer, 2016-02-17:
Hi @uli99 , 
do you try to write '/dev/rfcomm0' directly in the textbox?

[image]

vlachoudis, 2016-02-18:
Probably is better to be reported on the pyserial to add it in the port_list

closed 2016-02-18 by vlachoudis