vlachoudis/bCNC #243: AutoLevel probe direction is wrong

flaplante, 2016-02-19:
I am attempting to add auto level to my PCB milling process. CNC machine has been running fine except for the lack of a flat surface. So auto-level seemed the best solution.

I have followed all the steps in the WIKI, but when I click on 'SCAN', the spindle moves upward until it hits the home/limit switch. At that point motion stops completely, not like a homing cycle.

I have Zmin set to -10 (default), and Zmax set to +10 (was originaly +5, the default).

I thought to try a simple Probe, but that failed because I could not understand what to enter for the coordinate values that were needed.

I am using GRBL 0.9j, and bCNC 0.9.1 pulled today just before this set of tests.

I can find nothing in any documentation that would let me control the direction of probing.

Fred LaPlante

lalo-uy, 2016-02-19:
Zmax is to high. If the surface error is less than 5 mm =>
Set Z=0 over the pcb, in the center or at a corner. Then setZmax a few millimeters above the pcb (3) and Zmin a few bellow ( -3) Too long travel slows the process also.

flaplante, 2016-02-19:
Thanks, I will try it.
If works I will modify the WIKI to match.

Fred LaPlante, WA1DLZ

flaplante, 2016-02-19:
Did not work. As soon as I click on 'Scan', spindle heads up to he limit 
switch.

Just to be sure I understand here is what I am doing:
1. Home machine to set MCS to 0,0,0 (back right corner, Z at upper limit)
2. Jog spindle to front left corner, tool touching board, set WCS to 0,0,0
3. Use AutoLevel - Margins to set work space limits.
4. Set Auto-Level steps (N) to 5, Z-min to -3, z-max to +3
5. Click 'SCAN' icon

At this point spindle moves up, and up, and up, .....

Yes, I do have machine set to use mm as default.

Fred LaPlante, WA1DLZ

vlachoudis, 2016-02-19:
Maybe it is just a typo, do you set the "MCS" or "WCS" to 0,0,0?
Also, could you look in the "Terminal" tab if the commands send to Grbl are correct?

Durga000, 2016-02-19:
When I am doing auto-level, I do not set Z=0 until auto-level done.
Just homing axes ( X:min; Y: min; Z: Max ) 
set WCS to X=0 Y=0 and then launch autolevel ( you can use  lalo-uy recommendations )
Then only move to known coordinate (inside the probed section) for setting the Z=0.
My probe is connected to the Zmin pin ( I'am using ramps), is yours configuration the same ?
Is single point probe working ?

flaplante, 2016-02-19:
After homing the machine, and then moving the spindle to the near left corner of the work I have the following:
WPos    X=33.86       Y=10.0         Z=5.62
MPos    X=-183.69    Y=-129.85    Z=-37.805

After clicking 'Margins' I have
AutoLevel  
Xmin=0.0    Xmax=71.12    step=17.78    N=5
Ymin=0.0    Ymax=25.4      step=6.35     N=5

After clicking 'SCAN' the spindle moves up until I stop it. at that time the terminal window shows
G90 G20
ok
G0 Z3
G0X0Y0
G38.2 Z-3 F10
ok
G0 Z3
G0 X17.78 Y0
G38.2 Z-3 F10
ok
.
.
.

And No, the single point probe has same effect, spindle heads up and up and up...

closed 2016-02-19 by flaplante

vlachoudis, 2016-02-19:
@flaplante it is the G20 that causes the problem. You are switching to inches. Most probably you have in the startup commands a G20

flaplante, 2016-02-19:
That was indeed the problem. 
Thanks for the help.