vlachoudis/bCNC #244: Num pad buttons for jogging

twforeman, 2016-02-19:
This is an enhancement request.

Using the num pad for jogging works very well, but there is one more thing that could be useful.

The jog pad in the GUI has the "O" button in the middle to execute a macro (send the machine "home"). There does not appear to be a num pad button mapped to this button. I would suggest either "5" (since it's in the middle of the pad like the "O" button) or the "Home" button, for obvious reasons.

I would attempt the code modification myself but my python is terrible and I can't find the place in the code that maps key presses to movement. If you point me in the right direction I can make a pull request.

vlachoudis, 2016-02-19:
Thanks for pointing out. I will add the mapping to the home button

vlachoudis, 2016-03-05:
@twforeman I've added binding to "Home" buttons calling the "Home" routine and "End" button going to the origin. I cannot bind to "KP_5" since this is used as the literally for the number 5.

closed 2016-03-05 by vlachoudis