vlachoudis/bCNC #245: Unit conversion problem after using AutoLevel

flaplante, 2016-02-20:
bCNC doesn't seem to like my inch-based gcode inputs
the first block of my gcode input looks like this in the editor:
(move to start) g00 x0.20 y0.40
(bit down) g01 z-0.010 F50
(left) g01 x0.20 y0.60 F200
(top) g01 x2.60 y0.60
(right) g01 x2.60 y0.40
(bottom) g01 x0.20 y0.40
(moving height) g00 z+0.300

after auto level scan and running, the the terminal window has this:
G54G17G94g90g20
F300
G00Z0.3
g00x0.2y0.4
G1X5.08Y10.16Z-0.193656F50
 G1X5.08Y12.7Z-0.189728F200
G1X5.08Y15.24Z-0.191371
ok
 G1X14.224Y15.24Z-0.120014
ok
 G1X28.448Y15.24Z-0.009014
ok
ok
ok

It appears that auto level is converting my input from inches to mm. But it does not then remove the G20 from the GRBL feed. So all movements are too large by a factor of 25.4.  I tried removing the G20 from my input, but then all my gcode input values are treated as mm which are 25.4 times too small.

Any suggestions?

vlachoudis, 2016-02-20:
For the moment Autolevel expects that the units are not changed during run, and they have to be the same on what grbl is reporting. I would like to fix that in the next releases.
So for the moment what you can do is configure your grbl to report inches instead of mm (and bCNC as well) and run the autolevel in inches

atrueresistance, 2016-02-20:
@vlachoudis Would this happen to be the same problem I'm having?

After autolevel the machine goes to first point and then starts going at a 45 degree angle towards home.  

The following line is never followed up with an ok
[top.txt](https://github.com/vlachoudis/bCNC/files/139204/top.txt)

```
ok
G1X31.82366Y27.55392Z-0.509924
```

vlachoudis, 2016-02-20:
@atrueresistance if the gcode is chaning the units, then it could be that this is the problem.

atrueresistance, 2016-02-20:
@vlachoudis Nope, I'll open a new issue

flaplante, 2016-02-20:
@vlachoudis Per your suggestion to @atrueresistance, I pulled the latest code and can confirm that the G20 is pulled from the stream.
However, the inch to mm conversion is skipping the first few lines of the gcode
G54G17G94g90
F300
G00Z0.3
g00x0.2y0.4
G1X5.08Y10.16Z-0.291206F50
 G1X5.08Y12.7Z-0.307206F200
ok
As a result, the z0.3 which originally moved the spindle well above the board, now does not so that all following depths are WAY too deep, and the cutting does not start at the right place.  After that X & Y are fine, but Z is NOT.

vlachoudis, 2016-02-20:
@flaplante in the new master I am correcting for the unit change in the gcode. Could you please check.

flaplante, 2016-02-20:
@vlachoudis  Works great!!!
Thanks for the quick response.

closed 2016-02-20 by flaplante

vlachoudis, 2016-02-21:
Perfect!