vlachoudis/bCNC #247: Autolevel transform gCode not working correctly

atrueresistance, 2016-02-20:
After running the autolevel the following gcode file will not run correctly. The spindle makes it to 

`G00Z0.1
ok
G00X1.2541Y1.0788
G1X31.85414Y27.40152Z-0.487366
ok
G1X31.82366Y27.55392Z-0.488889`

but no ok is ever given. Then the spindle goes towards home. You can see in the image where the spindle is relative to the board. 

[top.txt](https://github.com/vlachoudis/bCNC/files/139224/top.txt)
[image]

Without the autolevel the code runs fine. I used flatcam to generate the code.

vlachoudis, 2016-02-20:
Could you change grbl to report inches, as well bCNC to think in inches and try again?

vlachoudis, 2016-02-20:
The first line in top.txt is G20 and is switching to inches.
If I find time the next days I will work to properly take into account the change of units in the autolevel part.

vlachoudis, 2016-02-20:
I've just pushed a modified version of CNC.py which in principle should expand the paths first with the correct units and then strip the G20/G21 from sending to grbl. I don't have access to my machine to check
could you check if it works with your code (of course without changing to inches the grbl reporting)

atrueresistance, 2016-02-20:
Yes, I am out right now. After running errands, I'll pull master and attempt again give me a couple hours.

vlachoudis, 2016-02-20:
Ok. I think I have a solution right now. The new master is correcting now the autoleveled path to reflect the units the gcode is setting. I still haven't tried on my machine but dumping the code now seems ok.

atrueresistance, 2016-02-21:
Wierd. So the first two holes it goes to drill them, and then goes back to origin. Not following the path like intended, but still cutting them where they are supposed to be. 

[image]

You can see in the etched board where it went back to origin and then to the point engrave directly. The etchings near the holes are a good example. I could see this behavior as a really good way to break bits. 
[image]

Autolevel Time
[image]

Autolevel really tacks on time compared to no autolevel

No Autolevel Time
[image]

vlachoudis, 2016-02-21:
I believe you have tested the first correction I've uploaded which was
changing the units to be similar with what grbl is reporting. However it
was not correcting the feed rate and the g0 motions :(

The second correction, is changing the autoleved path to reflect
the units in the gcode.
I've just run and dump your gcode with your probe information and here is the output
[run.txt](https://github.com/vlachoudis/bCNC/files/139505/run.txt)
Comparing the original (top.txt) and the autoleveled one the only
difference you will see is the z-correction in all the g1 motions.

atrueresistance, 2016-02-22:
@vlachoudis most likely. I'll try the board again as soon as Amazon delivers another PCB. So is the feedrate and g0 motions fixed or has this turned into an enhancement item? 
How does autolevel handle clearance? So for instance flatcam sets .1 as the default safety height. If autolevel finds a valley that is .2 down followed by a hill that is +.2, does autolevel compensate for this?

vlachoudis, 2016-02-22:
G0 and feed are fixed. Autolevel is only applied to the g1,g2,g3 so clearance has to be higher than any position of your pcb

atrueresistance, 2016-02-24:
autoleveled a few jobs yesterday. looks like feed is good. closing issue

closed 2016-02-24 by atrueresistance

vlachoudis, 2016-02-24:
perfect and sorry for your PCB. I will tag this as a new release

atrueresistance, 2016-02-26:
Working pefectly thank you! [image]

vlachoudis, 2016-02-26:
Nice. Why you don't make multipass? I use pcbgcode for eagle and with 2-3 multipasses it makes much nicer separation and cleans all the burr.

atrueresistance, 2016-02-26:
Mostly because this is the first one ever, I've designed and made. Boy was it a learning process. For passes what was your depth per pass and feed?

vlachoudis, 2016-02-27:
@atrueresistance  I didn't mean multipass in depth, but in the outline to make the isolation of traces thicker. I use a V-carve bit with 0.1mm tip and 30deg angle
and I typically ask 0.4mm isolation so it would make 4 passes.
Depth depends on the pcb. On the cheap ones I use 80um one some good ones I use 150um
Spindle is a 12000rpm and feed rate of 500-700 mm/min