vlachoudis/bCNC #248: Jog Interface Units always set to MM

twforeman, 2016-02-24:
Whenever the machine is homed or reset the jog units reset to MM.

I set Grbl to report inches, and that makes bCNC show inches in the DRO, but the jog units stays MM until I change it.

I tried setting "units = 1" in the user file in both the [control] and [cnc] section, but it doesn't make any difference.

Is this currently user settable? If not, can it be?

Thanks!

vlachoudis, 2016-02-24:
Do you have any startup commands in grbl that will change the units?

twforeman, 2016-02-25:
Ah. I did not, but setting $N0=G20 made the units stay inches.

Great!

closed 2016-02-25 by twforeman

vlachoudis, 2016-02-25:
I vaguely remember that @chamnit once said that the default units in grbl are mm and after reset is switching to it.

chamnit, 2016-02-29:
@vlachoudis : Grbl defaults to millimeters, but users can actively change this with the startup strings ($N0=G20), which are executed upon a clean reset (not after alarm).

neomadic, 2018-06-08:
Sorry to resurrect this but it's probably unworthy of a new thread.  Is there any way to get this to default to inches or throw some kind of alarm when it resets?  I wouldn't even care if it crashed the entire program. It seems like it resets to mm randomly between jogs, even when running gcode sometimes if I have to pause and restart, then everything's all screwed up.