vlachoudis/bCNC #249: Probing user defined paths

MForlong, 2016-02-24:
Hello!

I'm sorry if this posted in the wrong area.

Thank you very much for this awesome software!  I was looking over the probing function and I was wondering what would be the best method (if any) to probe say the points along a circle that might be distorted. Instead of probing 1000 irrelevant points on a grid? I have a few simple but relatively large parts that only need to be flat in small areas. It would be nice to only probe the areas that are of interest to me.  Would this be done in a macro or plugin?

Effer, 2016-02-24:
Hi @MForlong ,
I proposed more or less the same here [(https://github.com/vlachoudis/bCNC/issues/29#issuecomment-100668436)]
Then I forgot till now :) 
Maybe I can have a look and see if something can be done.

MForlong, 2016-02-24:
I was thinking the WCS, "Z mapping of a desired area for possible deformation in the z" I was wondering how hard it would be to define the "desired area."

Ozzy62, 2016-02-24:
It is possible to control the area that are used for probing in the probe set up.

MForlong, 2016-02-24:
Hi Ozzy,

Can it for example: I have a part 100 mm x 100 mm and I need 1 mm resolution only at certain points (maybe 20).  I would like to only have to find and store the Z offsets of the small areas of interest instead of having to prob 9,999 points on a grid.

Ozzy62, 2016-02-24:
What you can do is restrict the area that you probe, and set the number of point in each direction. But they will be spaced by the same distance, so no I don't see a way of probing individual points in individual areas.

lalo-uy, 2016-02-24:
The question is what happend if I probe between (10,10) and (20,15) and with that map mill from (1,1) to (100,100)

Effer, 2016-02-24:
Only the probed area is altered, all the outside area is streamed as it is.
My propose was to automatic avoid probing area where there's no gcode.

vlachoudis, 2016-02-29:
I think @Effer proposal is the easiest. Probe only the locations that are around the gcode. Most of the code for doing it is already present. I will think if there is an easy way of calculating it without extensive intersection calculations which are expensive in python.

Effer, 2016-02-29:
@vlachoudis I give a look to the code but it's too complex for me, it need many modifications to the codebase and I prefer avoid mess up the code.
Do you think a graphical approximation approach could be possible: I mean using the canvas rendering to determinate if a sub-region is not full white.

vlachoudis, 2016-02-29:
@Effer indeed a a graphical approximation from the canvas can be a solution.