vlachoudis/bCNC #250: bcnc crashes with ubuntu 14.04 an latest update

theblackraven, 2016-02-27:
hi,

i used bcnc without problems. Yesterday i updated ubuntu 14.04 and bCNC.
When i begin to mill Xorg crashes everytime (Xorg crashed with SIGABRT in __libc_message()
So i made a clean install with upgrade and had the same issue. Nothing other except antimicro (for using my cnc with an gamepad) is installed.

vlachoudis, 2016-02-27:
What graphics card you have?

theblackraven, 2016-02-28:
I have a thinkpad x201 for milling. The graphics card is a Intel GMA HD (5700MHD). 
By the way. Really great work.

vlachoudis, 2016-02-28:
I think this should be a bug of xorg or tkinter for the specific video driver. The problem is I cannot reproduce on my laptop to debug it.
Could you try to turn off some of the drawing features, like, workspace, grid, probe image, cut paths, etc...
Check also when does the crash of xorg happens, during zooming, panning, cutting etc...
BTW do you get a coredump from xorg, to open a bugreport with them.

theblackraven, 2016-02-28:
It seems, that this issue is caused by xorg.

I downgraded xorg with following command:

sudo apt-get install xserver-xorg-lts-utopic libqt5gui5 libgles1-mesa-lts-utopic libgles2-mesa-lts-utopic libgl1-mesa-glx-lts-utopic libgl1-mesa-glx-lts-utopic:i386 libglapi-mesa-lts-utopic:i386 libegl1-mesa-drivers-lts-utopic

After that and a reboot everything seems to work fine.

The crash happened everytime at the beginning of milling.

vlachoudis, 2016-02-28:
Ok, thank for letting us know. I am closing the issue.

closed 2016-02-28 by vlachoudis

spanner888, 2016-05-12:
First responses to your queries, then a lot of updated info:-
  Example code uploaded takes 90 seconds.
  Lock up persists during AND LONG after. Note cannot alt tab etc, keyboard lock light does not toggle, Linux console keys don't work (ie Ctl Alt F1-7) etc.
  CNC will only run short time then stop, eg 2 circle cuts of a 15mm bowl. Presumably this is just gcode in grbl buffer.

  Only when stepper enabled - hmm either that was poor testing/observation by me, or another change in behaviour. Now get crash without steppers or spindle running.
  Note crash is different behaviour to lockup originaly descrbed. See below.

  Not directly related to Intel card video problem #250, but maybe related via xorg, again more info below.
  
Updated testing information:-  
  - Problem has shifted from lockup to an "instant logout", which is probably an xorg crash - sometimes get a crash report for xorg appear on logging in again.
  This even occured on computer that is not internet connected, so no idea why the behaviour changed.
  
  - Problem occurs on non-Intel video cards - have reproduced on three different nVidia cards, in two computers. 
  
  - Same computer running Lubuntu 15.10 or Windows 7 is *OK*, but running Lubuntu 15.04 *crashes*.
      The newer Lubuntu info is:-  
	Linux 4.2.0-16-generic (x86_64)
	Distribution	Ubuntu 15.10
	Desktop Environment	LXDE (Lubuntu)
	...
	Loaded Modules
	  nouveau	nVidia Riva/TNT/GeForce/Quadro/Tesla
	  video	ACPI Video Driver

      The older Lubuntu info is:-  
	Linux 3.19.0-15-generic (i686)
	Compiled	#15-Ubuntu SMP Thu Apr 16 23:32:01 UTC 2015
	Distribution	Ubuntu 15.04
	Desktop Environment	LXDE (Lubuntu)
	...
	Loaded Modules
	  nouveau	nVidia Riva/TNT/GeForce/Quadro/Tesla
	  video	ACPI Video Driver  
  
  - Problem *DID NOT* occur on a old Lennovo model 9088ABM with Video card **Intel** Q33/Q35/G running OpenSuse 4.0.5.3, so it is not an issue with "every" Intel video card.
  
  - Problem *DID NOT* occur on exact same hardware & software configuration, except running bCNC 0.5, instead of bCNC  0.9.3 - when problem DOES occur.
    Verified this on two of the above computers.
  
Why so much Lubuntu? Well the two computers that have been connected to CNC are really old - I got as extra unwanted items in pallet load of IT gear from an auction.
So using Lubuntu as it is supposed to work on older & low resource systems.

Novena desktop (see https://www.crowdsupply.com/sutajio-kosagi/novena/) running Debian 8 is OK.

Attached is xorg crash information from the Lubuntu/Windows computer: [2016-05-12-034825_1513x1018_scrot.png.zip](https://github.com/vlachoudis/bCNC/files/260830/2016-05-12-034825_1513x1018_scrot.png.zip)

Can provide more info, but probably best if someone can push me in correct direction, else I will flounder around....

Thanks for looking at this and especially for bCNC.

spanner888, 2016-05-12:
oh dear - pasted into wrong tab/issue - please refer to issue 315 instead of here.