vlachoudis/bCNC #251: Stepper controlled by pure GPIOs for RaspberryPi?

Acer54, 2016-02-27:
Hello, 
hopefully this question is not too stupid. At the moment, I try to build my own little pen-plotter with the raspberry pi, 2 h-bridges and a selfmade mechanic utilising 2 steppers and a simple servo-motor.

Is there any "realistic" chance to use this (very good) programm? Maybe the gcode can be "forwarded" to a callback of a third python class or so? 

Do you think that this is possible without working through your whole lot of work? (I´d like to your the g-code editor at least... maybe I can port the editor to a PyQt4 app ?)

vlachoudis, 2016-02-28:
I saw somewhere how to make a file/pipe in linux to appear as a serial port. Then you can redirect all the commands there and you can handle it from another program reading from this file.

Converting to PyQt4 could be a lot of work, a big fraction of the program is based on tkinter.

lalo-uy, 2016-02-28:
Pulsing the stepers frrom user code will never be fast ans esteady on linux.
You shuld consider doing a device driver that simulate a stepper control.
Without current limiting speed will always be slow.

Acer54, 2016-02-29:
thanks for your Feedback and Response!
In the "raspberry-pi" community different "standard-projects" are existing (basically discribed with some YouTube Videos and some tutorials out there in the web). So I try to establish a software-base for these kind of projects based on Python, some webinterface and maybe with a GUI based on PyQt4 (without Special reason... just because this is the only GUI-Framework where I have collected some experiance until now).
The Intention was, to take bCNC and transfere the GBRL Transmission to anything which is utiliesing the GPIOs which are Controlling the Steppers directelly using a simple "h-Bridge" (which is available for some bucks on ebay). Most of these guys (including me) dont want to built high perfoming mechanics or controllers... they just want to move a pen, a knife or a Laser to a specific x/y point.... 
But bCNC seems like very very complex until now.... so without having to make big changes and work through the logics of this code, I think it is easier to take some features which I am searching for and implement it in a new Programm which is cut to the very basics of a pen-plotter (no z-axis for example... I only use "on" and "off" given by a servo-motor).
@vlachoudis I´m very interested in providing some kind of g-code visualisation and editing... can you give me a hint, where to find the functions which are performing the different renderings and manipulations to a given g-code which bCNC is providing? Am I allowed to use those? (= non-Commercial of course)
Thanks in advance, Matthias

chamnit, 2016-02-29:
@Acer54 : In short, if it were possible or a good idea to have a RPi output the GPIO for steppers or an h-bridge, it would have been done by now. The main problem is that Linux, or any major OS for that matter, is a real-time OS. GPIO timing is not guaranteed. This leads to steppers and motor controllers to have jitter and that jitter causes high-acceleration impulses that can lead to lost steps/motor stalling. 

The reason Grbl is on a micro-controller is to ensure that the stepper output is always guaranteed and has very low jitter.

Acer54, 2016-03-02:
I see, thanks. So Issue is closed, because there is no Intention to implement GPIO-Steering of h-bridges, because it does not make sense. For my needs doing like this is anyway enough, so I´ll wirte a simple program by myself. Thanks anyway for supprt and advice.

closed 2016-03-02 by Acer54

beta-tester, 2016-04-12:
how is LinuxCNC doing it? there you can drive the stepper driver by the parallel port. i mean handling a PC's parallel port pins is even worse than handling GPIO pins...
why is under LinuxCNC not that limitation (high level OS -> low level pin driving)

vlachoudis, 2016-04-12:
@beta-tester I have read somewhere that linuxcnc requires a special kernel with the RealTime extensions.

beta-tester, 2016-04-12:
ok, that makes sense then... i wasn't thinking about it.