vlachoudis/bCNC #252: Pressing stop button while running program resets WCS to G54

twforeman, 2016-03-01:
If I press the stop button in the middle of streaming gcode the work coordinate system gets reset to G54.

If I'm not paying attention then when I start the program again it will be using the wrong WCS.

It would be safer if it did not reset the WCS to G54 when you press the stop button.

vlachoudis, 2016-03-03:
@twforeman thanks for pointing out. When the Stop button is pressed bCNC is sending a feed hold to halt the machine in a controlled way and a soft grbl-reset to clear up the buffer of grbl. It is not bCNC that is resetting the WCS to 54, but when grbl is reset it initializes the WCS to 54.
For the specific purpose of the "Stop" button I agree it is safer to remember and restore the WCS after it is reset by grbl. I will do it. Also now that you are mentioning maybe some other values have to be restored like the TLO

vlachoudis, 2016-03-05:
@twforeman in the new master I am saving and restoring the G state as well the TLO after a Stop, but not after the "Reset" button. This I leave to grbl.

closed 2016-03-05 by vlachoudis