vlachoudis/bCNC #254: Manual tool change TLO

eduardsik, 2016-03-03:
Hello,

I cant realize if Im doing something wrong or the tool change not working properly. Please take a look at the description below of the process.

i set up the probe and is working good and set the tool change, probe locations. 

Calibrating the tool:
- put a new tool in collet, send the machine to home. 
- leveling the tool with a plate on the surface of the workpiece, using probing
- press calibration and calibrating the tool
- testing to return to the 0 point
- click on toolchange, the ganrty moving in the change position. 
- changing the tool 
- press pause, then moving to the probing position and measuring the offset
- moving to the change positon, press pause
- now the tool returning to the prewious position. If the new tool shortest than the previous has the gap to the workpiece exactly to the difference as the previous tool. If longest.. crash. So, it seems not measuring the offset or Im doing something wrong.

I appreciate your help.

vlachoudis, 2016-03-03:
The procedure is correct. Could you try with the other policy WCS. maybe something is resetting the tool offset

eduardsik, 2016-03-03:
Thanks for the replay. I updated the repo on my raspberry and now works
good.

eduardsik, 2016-03-03:
Sorry, response :)

vlachoudis, 2016-03-04:
OK thanks

closed 2016-03-04 by vlachoudis