vlachoudis/bCNC #256: Modal Group Errors - Fusion 360

mecizeldec, 2016-03-04:
I'm getting this constantly.

Hardware: RASPI 2, Protoneer CNC Hat (GRBL on Nano)
Software: BCNC (Raspi Dist.), Fusion 360

I've been beating my head against the wall (and google, forums and anything else even remotely similar) for two days over what post processor to use, as anything that seems to make sense keeps throwing these Modal Group Errors. Does anyone know offhand how to get past this with Fusion 360? First Homebuilt CNC and i'm soooo close to my first cut.

I've consulted the GRBL group and the code works fine for them... 
Is there a setting I messed up somewhere?
Do I need to use a different Post Processor? I'm using the generic GRBL post processor with Fusion.
[tEST bOARD.zip](https://github.com/vlachoudis/bCNC/files/158068/tEST.bOARD.zip)

Thanks!

vlachoudis, 2016-03-05:
Modal error is when you have two conflicting motion commands in the same line.
However your file runs happily with bCNC and grbl 0.9j.
The only thing is the tool change that is requested. T1 M6 commands.
In bCNC you have several ways of addressing these commands, the default is to ignore them
or you can setup your machine for a manual tool change with a probing location. Look the "Probe->Tool->Policy"

Now your modal group error I believe it might come from some error settings in grbl $Nxx commands or some wrong configuration in the "startup" commands of bCNC (Tools->Config->Machine->Startup)

vlachoudis, 2016-03-13:
@mecizeldec Did you found the problem?

mecizeldec, 2016-03-14:
Thabks for following up! I'll tell you in a few days when my replacement board comes in. The modal error sometimes happened at different spots when run consecutivelyt. I've witnessed some other strange behavior with jogging x any y (missed steps and such), so I ordered a replacement board. I suspect I damaged the logic translator when i shorted out one of the power supplies a while ago. It aall worked but I think there's a pile of noise on the lines. I'll keep you posted.

mecizeldec, 2016-03-26:
OK, new board installed - Check
Settings updated - Check

Still getting modal errors right off the hop....

I'm the only person seeing this, what am I doing wrong?

vlachoudis, 2016-03-26:
Could you please send us your config file
Tools -> Config -> User File

mecizeldec, 2016-03-26:
Under tools - Config I only have 4 options. 
Machine
Fonts
Colors
Shortcuts

vlachoudis, 2016-03-26:
In the label "Config" there is a small black down pointing triangle. This opens the menu with the additional options

mecizeldec, 2016-03-26:
Not on my screen...
[image]

mecizeldec, 2016-03-26:
[Chris Config.zip](https://github.com/vlachoudis/bCNC/files/190661/Chris.Config.zip)

vlachoudis, 2016-04-20:
The problem was the "move" command that was by mistake converting the "G28 G91" to "G91 G91" which was causing the Modal error. Now it is correct.

closed 2016-04-21 by vlachoudis