vlachoudis/bCNC #257: End-stop

cyryllo, 2016-03-04:
I use a program with Cyclone PCB. I have two endstop for the x-axis and y-axis. I don't see in the possibilities use zero position using these end-stop.

CarlosGS, 2016-03-04:
Hi @cyryllo, sorry but I don't get what you mean :S
Can you give us some more detail?

closed 2016-04-07 by vlachoudis

cyryllo, 2016-04-08:
Hello
I try set home in program, but get error
[image]
I setup my cyclonePCB https://github.com/carlosgs/grblForCyclone/wiki/1%29-Configure-bCNC-for-Cyclone
next try home 
[image]

https://github.com/carlosgs/grblForCyclone/wiki/2%29-Hello-World!-Mill-a-bowl-with-bCNC

Effer, 2016-04-08:
Hi @cyryllo , could be that the homing is disabled in GRBL? Can you try to read parameter $$?

[image]

Sorry for this terrible screenshoot

cyryllo, 2016-04-08:
Yes in cnc shield and Uno i have $22 disable. Now upload on ramps and mega and have enabled. Where can I change this in uno and shield? I use this firmware https://github.com/carlosgs/grblForCyclone/tree/grblForCyclone_CNCShield

cyryllo, 2016-04-08:
I was turned on in the settings. Just make it work I had to clear the memory EEPROM :).
Now i enable ;)

CarlosGS, 2016-04-09:
Awesome!!
I've added a note in https://github.com/CarlosGS/grblForCyclone/wiki/1%29-Configure-bCNC-for-Cyclone
Remember that you can also edit the github wiki pages if you find any step that is confusing, in order to help other people :)
Thanks @Effer for helping to find the issue