vlachoudis/bCNC #258: Psuedo terminal or Ethernet connection

bigguiness, 2016-03-04:
Hello all,

Is it possible to get bCNC to connect to a controller over a Psuedo terminal port or an Ethernet connection?

I have a controller that uses an Ethernet socket for its connection. Normally I use telnet or socat to send data to it.

I have tried using socat to relay data from a Psuedo terminal to the controller and it works from the command line:

# from one command line, create the psuedo terminal (/dev/pts/3) <->  TCP connection
$ socat PTY TCP:<address>:<port>

# from another command line, send some data to the controller
$ echo "g28\n" > /dev/pts/3

This works and my machine does a homing sequence.

I then tried using /dev/pts/3 as the Serial Port in bCNC and get an "Error opening serial" message that says:

could not open port /dev/pts/3: [Errno 2] No such file or directory: '/dev/pts/3'

Is something like this possible?

Regards

lalo-uy, 2016-03-04:
May be a permisions problem.
Try running bCNC with sudo.

bigguiness, 2016-03-07:
Running with sudo doesn't work.

$ sudo ./bCNC.py
No protocol specified
Traceback (most recent call last):
  File "./bCNC.py", line 2225, in <module>
    tk = Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1712, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"

vlachoudis, 2016-03-08:
Running as "sudo" doesn't allow you to start the "X11" application. It is better to change the permissions of the file like "chmod 777 filename"

vlachoudis, 2016-05-04:
Did you succeed?

bigguiness, 2016-07-29:
I just tried pulled bCNC again and tried it.

Looks like I get a connection now but my controller does not like the commands from bCNC.

I hooked up the TCP socket to a PTY with socat.
$ socat PTY TCP:192.168.0.101:4444

Then started bCNC and setup the Serial Port for /dev/pts/3 @ 115200 baud.

The Status line still shows "Not conencted" but clicking on it says that a "Connection is established with Grbl".

But when I try to enter a command in the terminal I get error messages back from the controller.

Entered command: G28
Response:
'.
// Line 1351 G-Code Syntax Error: Letter '?' is not followed by a number but '???????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????G28
'.
// Line 1352 G-Code Syntax Error: Letter '$' is not followed by a number but 'G
'.

The G28 is there but all the ? are messing up the parser.

Harvie, 2018-10-29:
There is software available that allows you to access serial port on remote machines over IP.

closed 2018-10-29 by Harvie