vlachoudis/bCNC #259: Connection win 7 64 bit

milezdr, 2016-03-05:
Hi. I just find out about this grbl GUI and I have hard time connecting my arduino to the bCNC.
I'm using win 7 x64bit and I get this errors 
[image]
[image]

vlachoudis, 2016-03-05:
Unfortunately the serial library is not part of the standard python package.
So as the dialog claims you have to install the additional serial library from python with the command that is mentioned there for windows

closed 2016-03-05 by vlachoudis

Effer, 2016-03-05:
Hi @milezdr , it seem that you are missing pySerial module. It's a library for python that take care of serial comunication. You can easy install it following the wiki instruction: 

"
use the correct installer for the same version of your python installation. From a terminal (cmd) you can execute the command (where PythonXX is your python installation directory)
C:\PythonXX\Scripts\easy_install pyserial
"