vlachoudis/bCNC #262: Windows 10 Python 2.7.9 bCNC.bat

rwensley, 2016-03-06:
Greetings,

I am looking for help in getting the bat file to run.  

When I click on it a screen flashes then disappears.  No errors can be seen as it happens to fast.
I have also tried 'Run as Administrator'

Any help will be appreciated.

Ron Wensley

Protoneer, 2016-03-06:
Insert "pause" at the end of the bat file to keep the cmd window open or
run the bat from cmd line.

motanescu, 2016-03-09:
Hi Ron,
I think i had a similar problem
https://github.com/vlachoudis/bCNC/issues/266

alfredanil, 2016-03-10:
+1 
Same error, not working from cmd line either

vlachoudis, 2016-03-10:
Can you post a copy of the output from the terminal.

closed 2016-04-07 by vlachoudis

mehadisaki, 2017-08-28:
I have same problem. cmd also show error

rasciodc, 2018-11-21:
+1
Same problem here!

Already tried python x86, x64. No luck.

I used bCNC in the past on same laptop and used to run from bat file. 

Put "pause" at the end of the bat but all it gives is a cmd window showing "press any key to continue..."

Harvie, 2018-11-21:
@rasciodc Do you have python2.7 installed?

rasciodc, 2018-11-21:
> @rasciodc Do you have python2.7 installed?

Yes. 2.7.15 x86.

Tried to follow steps from #266 with no success. (tried pyserial 3.4 and 3.1)

Harvie, 2018-11-21:
This is the best way to install and lunch bCNC on windows for now:
https://github.com/vlachoudis/bCNC/issues/964#issuecomment-440315914

It's also described in README

However it does not create any clickable icon. I don't use windows, so i don't really know how to do that. But i guess it will improve in the future. Pull requests welcome :-)

rasciodc, 2018-11-21:
> This is the best way to install and lunch bCNC on windows for now:
> [#964 (comment)](https://github.com/vlachoudis/bCNC/issues/964#issuecomment-440315914)
> 
> It's also described in README
> 
> However it does not create any clickable icon. I don't use windows, so i don't really know how to do that. But i guess it will improve in the future. Pull requests welcome :-)

After following the instructions I'm able to run bCNC from the bat file again! So weird...

I observed that NumPy was installed during the process, which I'm not sure I had installed before... So I cannot confirm that it was the problem but could be related (as stated in #468).

Well done, thanks!!

Harvie, 2018-11-21:
> I observed that NumPy was installed during the process, which I'm not sure I had installed before... 

Yes. It's very likely the cause. Check the README next time :-)

rasciodc, 2018-11-21:
> > I observed that NumPy was installed during the process, which I'm not sure I had installed before...
> 
> Yes. It's very likely the cause. Check the README next time :-)

Yeah, I missed that... my bad. =)

But talking about user experience, a Windows user will inevitably follow steps on https://github.com/vlachoudis/bCNC/wiki/OS#windows , which makes it clear you need to install, in addition to Python itself, only the PySerial package. Perhaps adding NumPy to that list wouldn't hurt... 

Both the ReadMe and the link above mention the need for PySerial. Perhaps NumPy should also be mentioned in both places.

Thanks for your prompt and quick help!

Harvie, 2018-11-21:
Yes. I have to update the wiki. Pip will now install all packages automaticaly, so there's no need for user to install them manualy one by one.