vlachoudis/bCNC #265: Red Circle with Diameter of End Mill?

paranormalactivities, 2016-03-07:
Hello Vasilis,

first of all: Thanks for writing this great Software, which i'm using since i finished the first Version of my self made wooden cnc mill in the beginning of this year. 

for calibrating the workpiece on the machine, it would be nice, if the red gantry position marker would have the same diameter as the end mill i'm using, so selecting one of the end mills or editing the diameter directly would be fine. 

By the way: I wrote a tool, which loads the gcode file generated by bcnc, and converts the z-axis-moving to helical scan / smooth movement for selected blocks (which are containing an identifier in their block names). after that i reload the converted script into bcnc and start milling. Unfortunately i wrote it in Delphi, but if you would like to adapt the algorithm which converts it, or somebody else is interested, i would publish the source and/or the exe-files (Windows only). I tried to understand the bcnc-python source and to program it directly into bcnc, but i think i've to learn python first :-)

vlachoudis, 2016-03-08:
@paranormalactivities I think this is a good suggestion for the gantry.

Also if you want to convert it to python, we can add it as a plugin. It would be a good exercise for learning python :)

vlachoudis, 2016-03-27:
@paranormalactivities in the new master the gantry diameter is the same as the selected EndMill as you suggested.

closed 2016-04-07 by vlachoudis

Effer, 2016-04-08:
@vlachoudis  I like the tool size been in real scale. Wouldn't be nice to have it reflect also the EndMill geometry? Conical, cylindrical or spherical?

vlachoudis, 2016-04-09:
@Effer maybe some time in the future :)