vlachoudis/bCNC #266: Python flash but no GUI when running bCNC.bat on Windows 7

motanescu, 2016-03-07:
If you installed Python 2.7 x64 you need to uninstall it AND DELETE THIS FOLDER C:\Python27.

Next reinstall Python x86 and pySerial.

Tested with Python 2.7.1 x86 and pySerial 3.10 x86

vlachoudis, 2016-03-08:
Did you try to installing the pySerial for the 64bit version?

motanescu, 2016-03-09:
hi, I was actually posting a possible fix to the issue, maybe others have the same symptoms.

What I did:

- Installed python 2.7 x64
- Attempted to install pySerial but failed because pySerial cannot find python in the registry (known issue with 64 bit version)
- Uninstalled python 2.7 x64 (from control panel)
- installed 2.7 x86 and py serial successfuly
but
- at this point when running bCNC.bat the GUI would not come up. The python window appears for a split second and then disappears.
- I spent a full weekend trying to figure out why the bat file wouldn't run anymore
I fixed it this way:
- Uninstalled python, pyserial AND deleted C:\Python27 (this last step is key because the folder still has leftovers from the x64 installation causing the x86 version to crash)
- Reinstalled python 2.7.1 x86 and pyserial 3.1 
bCNC GUI loading ok, python stays up like it's supposed to
Hope I will spare some people the pain Cheers!

closed 2016-03-09 by motanescu

MrMurdjo, 2019-08-24:
While I managed to get bCNC to start with Python 2.7 in Windows XP (32 bit) and Windows 7 (64 bit) there are two issues:
1. Com port is opened but the state button stays red and does not update while clicking on it displays a message that a connection with GRBL is established. It is, because the real status is read from there. But jog buttons and manual commands seem to not send data as the LED on Arduino is not blinking.
2. Trying to open a .nc file opens a file load dialog but the dialog does not allow changing to another partition (drive letter). Stays only within the limits of current partition (c:) where the bCNC.bat and the rest of the files are copied. I modified the .BAT file to explicitly call Python 2.7 in Windows 7.
Under Ubuntu 64 bit it is working more or less as expected.

BlakusSenior, 2020-10-19:
Hi All,
I have the same problem so I'll try that fix. On the subject is there a version of the bCNC.bat that is understandable to some one like me with only a basic list of bat file commands ?
.....Blakus