vlachoudis/bCNC #267: Z drops to -14 after the first direction change  with autolevel

motanescu, 2016-03-09:
I am trying to engrave an aluminium piece. 
I import g-code generated in F-Engrave (V-carve function). The gcode runs as expected without autolevel, however horrible things happen after I probe  the surface and zero it out:
I probe the surface with autolevel (get margins from gcode, standard interval of 5)

The heightmap shows maximum +/-0.7mm after probing.
The start of the job is a rectangle, aproximately 90x50mm. the gcode shows a lowest Z value of -0.3. After the first direction change, however, the z level drops to -14 without a reasonable explanation. I have ruined a few jobs already. Any thoughts apreciated.

vlachoudis, 2016-03-09:
Could you send your gcode and the probe file to check.

closed 2016-03-12 by motanescu

vlachoudis, 2016-03-12:
@motanescu did you found the problem?

motanescu, 2016-03-12:
I made some changes to my setup and the issue appears to be gone. I couldn't say what fixed it... I used the latest version 0.9.1, changed microstepping from 32 to 16 and managed to run the job without any issues. Here are my current working settings, the F-engrave screenshots are in green
[image]

vlachoudis, 2016-03-12:
ok thanks