vlachoudis/bCNC #268: After home code does not run

atrueresistance, 2016-03-09:
I'm not sure if I'm the only person that has this issue. After an autolevel and zero I move the Z up one or two mm and turn the Dewalt 611 on. Then I hit home. If I just hit go on the file loaded, the machine goes to about entry point and stops almost like the infamous Shapeoko 3 serial disconnect. Now, after homing if I move it just one step in any X,Y, or Z direction and hit play the gcode runs fine. Nothing super crazy, just something I've noticed.

vlachoudis, 2016-03-13:
@atrueresistance noise problem?

atrueresistance, 2016-03-14:
For the disconnect one yes, but dont think it is relate to the homing. I'll have to make a video showing what I mean tonight.

atrueresistance, 2016-03-17:
Of course when trying to shoot the video last night I couldn't get it to happen.

closed 2016-03-17 by atrueresistance