vlachoudis/bCNC #269: crash at job start

se7en9057, 2016-03-13:
when starting a job using cam paths generated from flat cam for PCB milling BCNC crashes and python begins a massive memory spike. after reaching 5G i killed the process. attached is the nc i was attempting to run. 

[front.zip](https://github.com/vlachoudis/bCNC/files/171167/front.zip)

system specs. 

intel processor
windows vista x64

vlachoudis, 2016-03-13:
What python version you use?
I've tested it on linux and it works fine, with 37Mb a bit more than without anyfile.

se7en9057, 2016-03-13:
Python 2.7.10 (May 23 2015)

vlachoudis, 2016-03-13:
I have the same version on linux.
I will check on Windows later.

se7en9057, 2016-03-13:
ok. thank you . if there are any log dumps just let me know where to find them and i'll post them.

se7en9057, 2016-03-13:
looks like running it in mm fixed the issue.

vlachoudis, 2016-03-14:
Did you use the auto leveling?

se7en9057, 2016-03-14:
Yes. But when I got it to work I did the leveling in metric. So I'm thinking the imperial measurement was the issue.

vlachoudis, 2016-03-14:
Could you send save and send us the autolevel .probe file (from the imperial test) so I could try to debug it.

se7en9057, 2016-03-14:
sure i'll have to work on that tonight when i get home.

se7en9057, 2016-03-15:
[front.zip](https://github.com/vlachoudis/bCNC/files/174821/front.zip)

hopefully i did that right. i did an auto level and saved it. 

i think this is user error all around tho.

vlachoudis, 2016-03-16:
@se7en9057 you've send the gcode and not the probe information. You have to save as ".probe"

se7en9057, 2016-03-16:
sorry. still getting familial with the software.

vlachoudis, 2016-03-30:
Do you still experience the issue or it is solved?

closed 2016-04-07 by vlachoudis