vlachoudis/bCNC #270: Autolevel in Laser Mode

atrueresistance, 2016-03-17:
I've been approached to laser engrave something on a not flat surface. I've used bCNC autolevel with a spindle successfully before, but never attempted it on the laser.

@vlachoudis If in lasermode does autolevel transform the Z level of laser code?

My thought is to attach a metal rod offset of the laser, spray adhesive aluminum to object, and probe area. Take off rod and aluminum, move laser to an autolevel point and zero. 
The laser is turned on with M3/M5 and intensity set S0-10000

Would this work?

vlachoudis, 2016-03-17:
@atrueresistance it should work. By setting the laser mode when importing a dxf it converts the entry/exit gcode to m3/m5 all the rest remains unchanged. So if there is a variation in z or from autolevel it should correct it. But I didn't check.

atrueresistance, 2016-03-21:
It does not work, just got done testing and can upload video if you need. It looks like it does not send the S command with autolevel. The Z does in fact move up and down in coordination with the probe data, however laser intensity is always low

vlachoudis, 2016-03-21:
Ok I see, the autolevel algorithm strips the S### commands. I will fix it.

atrueresistance, 2016-03-21:
Yep, I'd does it in regular and laser mode if that helps.

vlachoudis, 2016-03-27:
@atrueresistance I found some time today to include the stripped commands. Could you check it?

atrueresistance, 2016-03-27:
@vlachoudis Thank you, I'll try it as soon as I'm home. Hope you are having a Happy Easter.

atrueresistance, 2016-03-29:
@vlachoudis worked like a charm. A local guy makes bamboo bicycles. I'll make $15 for this and am willing to pass it on to the bCNC developers. Is there a donation PayPal? 
[image]

closed 2016-03-29 by atrueresistance

vlachoudis, 2016-03-30:
@atrueresistance glad it worked! No, there is no donation paypal, maybe I could do some day :)