vlachoudis/bCNC #271: Crashes when trying to send code

auarhau, 2016-03-19:
I described the problem here: http://www.shapeoko.com/forum/viewtopic.php?f=7&t=7835
Please see that link for closer description and uploaded files etc.

Basically the program crashes when I try to send code (play button). Everything else seem to work, including scan macro, homing and jogging. Other g code senders work.

I tried and replicated this 100% of the time with two different computers. I even swapped out the gfx card in one of them and tried all the proprietary drivers. I tried this under linux mint and ubuntu lts fresh installs. It also crashes even if there is no power to the steppers. 

What happens in a crash is showed in my link on a video. Basically the machine logs out (or goes black) and the cnc stops. This happens the second I press the play button.

vlachoudis, 2016-03-19:
It seems to be related to xorg. Look the issue #250 and how he resolve it.

auarhau, 2016-03-19:
Thanks. That worked for me also! :) He also had a thinkpad (Lenovo laptop). Both my stationary PC and laptop are Lenovo also (Thinkcentre and Thinkpad). Perhaps not a coincidence? 

Glad I got it working. Now I will have fun playing with your software. Seems very promising!

vlachoudis, 2016-03-20:
Glad that it worked. Maybe you should open a bug report to xorg, so others could benefit as well.

closed 2016-03-20 by vlachoudis