vlachoudis/bCNC #272: Open gcode very large file, wait a long time to prepare

viewsat, 2016-03-22:
I opened this gcode test of 97 million lines to be tested,wait a long time to prepare.
http://www.picengrave.com/downloads/darklylabs/Roy%20Rogers%20Metric.zip

When I use the latest Universal Gcode Sender open this gcode test, wait 10 seconds to read gcode,

atrueresistance, 2016-03-22:
@viewsat I use bCNC for laser code all the time. Turn off the render
[image] and it will load in 20 seconds.

vlachoudis, 2016-03-22:
Python is not very fast as language, so rendering can be long. Also it is in my plans to optimize via threads the sender.

tuxun, 2016-04-11:
Hi.
@vlachoudis maybe bCNC should warn us between gcode file loading and render starting? The fact is I would like to see: "loading (in ram) done (in XXX millisecond), rendering the ram gcode (work in progress X% )" ;)

Effer, 2016-04-11:
I've to say that this is the most annoying limitation of bCNC. Hit start and then wait till everything is loaded, even if GRBL can buffer only very few lines of gcode.
I'm confident in @vlachoudis to take care of this, and if you need some Beta tester let me know.

vlachoudis, 2016-04-12:
The long waited feature by @Effer and requested by others as well @wolfmanjm is now there. The new master is queueing the commands immediately the moment you press "Run" and is processing the remaining in the background. So once you click "Run" the response should be immediate. Could you please test it.

Effer, 2016-04-12:
Wonderful, tonight will be a test night!

Effer, 2016-04-12:
First 50K lines done! (Scrap wood)

[image]

Great job @vlachoudis it starts in a flash and run smooth and without issue. It's a pleasure run the code without that annoying wait. Thank you!

The only drawback (I know it now will seem "fussy") is that soon after start is not possible to stop neither pause the job with the software button.

vlachoudis, 2016-04-13:
@Effer I afraid you are right. The same thread that does the preparation of the g-code is also watching the buttons, so as long the gcode is prepared the GUI is inactive. I thing I have one solution I will try it later today.

vlachoudis, 2016-04-14:
In the current master the "Stop" and "Pause" are active during submitting.
Also to make it more responsive I've put a timer of 5s in the drawing in canvas. So in case there is a huge file that takes more than 5s to draw it will interrupt the rendering. It is user configurable in the .bCNC ini file. However I have to put an easier configuration directly from the canvas. I am thinking of a dropdown combox that one can choose directly the rendering time from 1, 2, 5, 10, 20, 30, 60, infinite

Effer, 2016-04-14:
Sorry to report that the issue with bCNC resuming the job after a stop by itself, is still there.
The status bar report "Run ended" but bCNC keep on streaming. Also you can exit the programm without the usual warning.

Effer, 2016-04-14:
It seems to happen more often with big files. When hit stop I get this error:

[image]

Then after 5-6 seconds bCNC start streaming again by itself.

vlachoudis, 2016-04-20:
In linux where I am testing it works fine and I cannot reproduce this problem. Maybe it is related with windows.

Effer, 2016-04-28:
Any suggestion in how to debug? I can't reproduce everytime but is still there with latest version.

vlachoudis, 2016-04-28:
Can you try the latest version I think I found the probem

st0jko, 2016-04-30:
First let me state that I love this piece of software and would like to thank everyone working on it. 

Now about Effer's problem.. I just experienced the same thing. I hit stop and bCNC seemed to just jump to another piece of code and continue to work there. It actually ruined the piece but whatever.. I had to hit stop multiple times, because it seemed to just jump to another piece of code to work from. Like the code was broken into smaller chunks, because it was so large(130k lines), and hitting stop just stopped current chunk, but not other that follow.

Anyway.. hope this helps.

vlachoudis, 2016-04-30:
@st0jko in which OS you are running?

st0jko, 2016-04-30:
Windows 10 pro x64 and windows 7 ultimate x86

Effer, 2016-05-02:
@vlachoudis I can reproduce in both Win7 and WinXp.
From my test the stopRun method in Sender.py is executed, but not the checkStop method in bCNC.py.

vlachoudis, 2016-05-03:
@Effer @st0jko could you please try the new master. I think I found it this time.

Effer, 2016-05-03:
@vlachoudis first tests are succesfull. More test from @st0jko are wellcome.
Thanks!

vlachoudis, 2016-05-03:
@Effer my guess is that the os.sleep(1) in windows was blocking the background thread on windows.

Effer, 2016-05-03:
I was just asking for explanation in PM.. you read my mind again.
That's fully compatible with some of the strange output I've eneconter in my test.

vlachoudis, 2016-05-04:
I am pretty confident that the last commit fixed the problem. So I am closing the issue, unless if it happens again.

closed 2016-05-04 by vlachoudis

st0jko, 2016-05-07:
Can confirm. Working well now.

vlachoudis, 2016-05-07:
Thanks @st0jko