vlachoudis/bCNC #275: Auto Level - Ignore G2 G3 commands

woodworkerbob, 2016-03-29:
Suggestion: When applying the probed level data against the g-code just ignore the G2/G3 commands. I think this is what Chilipeppr does.

I'm starting to do inlay work and am concerned about the conversion to straight lines.
You could make this a user option.

Thanks,
Bob

vlachoudis, 2016-03-30:
Ignoring them would be a BIG mistake, the machine will move in one straight line from the start to end!
bCNC interpolates them with multiple straight segments (ONLY for the autolevel) where the accuracy of each segment is user defined, typically is 10um maximum sagitta (max difference in the arc segment from a straight line) (Tools->Machine->Plotting Arc Accuracy). In any case this is what internally grbl does for arcs. For wood working you cannot notice 10um maximum error in the interpolation. But as I said you could still reduce the error resulting in more line segments.

I was trying initially to convert them to tilted arcs (converting it to a helical movement), but this generates higher errors when the stock it too tilted (machining a cylindrical object).

vlachoudis, 2016-03-30:
Look the result on #270

woodworkerbob, 2016-03-30:
Thank you for the response.  You are correct.  I'm being too concerned.  
The inlay project is planned for corian, but at 10um, it will not be 
noticeable. Please ignore this request and don't think of me as too crazy.

On a personal note:
How can I help?
Resume: I was a programmer for 35 years, currently retired.  The last 
few years of my career, I maintain a system partially written in 
Python.  I know the basics of G code having written several G code 
programs and 3 post processors for Vectric Aspire.  I'm new to the world 
of community software development, but I think I can help.

WoodWorkerBob

Bob

vlachoudis, 2016-03-30:
If you are using inches you could reduce it since this value is on the units you are using 0.01 inches = 254um.

closed 2016-03-30 by vlachoudis