vlachoudis/bCNC #277: Probe Camera

DaveBrauns, 2016-03-31:
I just wanted to use my USB camera, but the button is grayed out. I already found out that I have to install opencv and PIL, but it is still grayed out.
I tried using opencv from the python console and saved an image taken from my cam, which worked as expected, so I think the installation is ok.
Did I miss anything? Any suggestions?

vlachoudis, 2016-03-31:
@DaveBrauns could you try from the console to start python and execute the following commands
```
import cv2
import PIL.Image
import PIL.ImageTk
```
To see which dependency is failing

DaveBrauns, 2016-04-01:
Thank you very much!
The ImageTk module was missing. Now it works like a charme.
Great work and great help what you are doing.

closed 2016-04-01 by DaveBrauns

vlachoudis, 2016-04-01:
Maybe it should be mentioned clearly in the wiki