vlachoudis/bCNC #281: Webcam support.

knobblyknees, 2016-04-03:
Hi, great bit of software, really intuitive. Just trying to get webcam working. Have imported numpy and cv2. Camera toggle on canvas is greyed out. How do I enable this?

Effer, 2016-04-03:
Hi @knobblyknees , try to have a look here #277, maybe even for you the ImageTk module is missing.
We could create a small summary of the module required and suggested, maybe a checklist inside the software.

knobblyknees, 2016-04-03:
Thanks Effer, PIL was missing. Sorted.

closed 2016-04-03 by knobblyknees

vlachoudis, 2016-04-03:
@Effer it could be a good idea to create such a checklist. There is only a check for the python-serial which is mandatory, without Tkinter it wont start anyhow, but I didn't put any message for PIL, OpenCv, ImageTk since those are optional. But a checklist could be nice