vlachoudis/bCNC #286: GUI for Tablet

k5beason, 2016-04-06:
Hello,

I'm running bCNC on a Chuwi Vi8 Windows Tablet. They sell for about 99.00 bucks making it a perfect candice to run bCNC.

I'd like to change the location and size of the Jogging Controls and create a simplified interface for use straight from the tablet.

I know the web interface provides some of the functionality but I've got the tablet directly connected to the controller and would like to remain within the program.

vlachoudis, 2016-04-07:
I wanted to put customizable fonts for everything, but I forgot :)
By increasing the fonts the button sizes will increase as well.

k5beason, 2016-04-08:
Could I move the Controls to replace the workspace view on the control pane? so they resize and take up the right pane?

vlachoudis, 2016-04-17:
You can switch the order of the  panes in the bCNC.py . as well the jogging buttons could be made to take up all available space and resize with the pane.

closed 2016-06-09 by vlachoudis