vlachoudis/bCNC #288: Button Position

Guenni75, 2016-04-09:
Hi.
I have a "Problem" with the 'move Gantry' and the 'WPOS' Button.
How can i hide or move one of them? 
You press the WPOS button quickly accidentally if you just want to move the gantry.

Or is there a way to customize the Toolbar?

vlachoudis, 2016-04-09:
There is no configuration of the location of these buttons. One has to change the code in the CNCCanvas.py. Your comment is true. I believe they don't fit there. The toolbar should be intendent only for manipulating the canvas. Maybe I can put them as two big buttons in the Control panel under the X=0, Y=0, Z=0 buttons?

Guenni75, 2016-04-12:
Just swapping the Ruler and the WPOS Button would be much better.

vlachoudis, 2016-04-14:
I've temporarily move them (just for testing) under the Control status on the left. Let me know your impressions. I have the feeling that is better there.

Guenni75, 2016-04-15:
They are better there, but still side by side!!
You press the WPOS button quickly accidentally if you just want to move the gantry.

vlachoudis, 2016-04-15:
Yes, but they are bigger :). I could put one below x and the other below z.

Guenni75, 2016-04-16:
This is what im talking about ;)

Guenni75, 2016-05-09:
I think the button fit there very good.

closed 2016-05-09 by Guenni75