vlachoudis/bCNC #290: Auto bed leveling: z0 prior to xy causes crashes

cyruschilds, 2016-04-09:
When autobed leveling the z movement precedes the x and y movements which can cause the machine to crash into the fixtures. If the machine moved to the beginning xy coordinates( set for the autobed leveling) before moving the z down, it would avoid this problem. Could you please let me know where to make the necessary changes to do this?
Thank you.

vlachoudis, 2016-04-12:
Adding first the XY movement and then the Z can also causes problems. I would see more safe to
add first a Z movement to the Zsafe, then the XY and then the Z to the probe starting location.

cyruschilds, 2016-04-12:
Ok but I use a tool offset switch so the auto probe is all in the negative
so if I forget to move the probe off the switch fist it crashes and breaks
the probe. I just need it to move xy first or have a move to g54 x0 y0 at a
safe height option

cyruschilds, 2016-04-12:
Sorry all I mean is it would be nice if the z safe travel  height was
different to the probe retract height.
bCNC is very cool thank you.

vlachoudis, 2016-04-12:
zsafe is defined in the "Stock" and is different from the "zmax" of the Probe

vlachoudis, 2016-04-12:
In the new master, I've added the Z[safe] as movement before any XY.
You have to declare the Zsafe in the stock

closed 2016-04-12 by vlachoudis

cyruschilds, 2016-04-12:
All you have done is changed the dialog box text.

My z safe is normally 10mm but but my scan is somewhere around -2.8mm so it
takes a long tome to do a grid of 200/200 (2 hours plus) if I leave the z
safe at 10mm.
Thank you anyway I just change the code myself.

vlachoudis, 2016-04-13:
You missed completely what I've done. I didn't change any dialog, the scan process now
does
```
G0 Z Z[safe]     (the zsafe is defined in Tools->Stock, and it is different for every stock)
G0 X[xmin] Y[ymin]  (go to the first point)
and then the scan starts
G0 Z[zmax]  (the limit you define in the autolevel probe)
G0 X.. Y..
G38. ... etc...
```
while before the first two lines didn't exit

cyruschilds, 2016-04-13:
Thank you. So sorry I did miss completely what you had done.