vlachoudis/bCNC #293: German translation

feistus, 2016-04-11:
Hi vlachoudis,
I'm doing some German translations right now and I stumbled upon the question which speech to use. In German there is the formal ("Sie") and the informal ("Du") speech. 
While the formal speech is used to be polite and respectful towards the user it also implies a certain distance and sometimes feels a little bit oldschool. 
The informal speech is more common nowadays in the the internet and feels friendlier for most of the younger folks, although it might appeal not as professional as the formal speech to some of the older users.
I prefer the informal speech but I think the choice is yours. ;)

Effer, 2016-04-11:
For the italian translation it has been taken a formal approach by @onekk . I agree with his choice.
When I'm in doubt I try to use the same words used by OS system (Windows in my case).
Maybe some suggestion can come even from @ThierryM and @CarlosGS in this way all translations will use the same approach.

CarlosGS, 2016-04-11:
The spanish translation also uses a formal approach. I think it is better, since it sounds OK for us spaniards, and is more natural for latin-american speakers
But every language is different :)

feistus, 2016-10-06:
Thank you guys, I'm going for formal approach as well then.

closed 2016-10-06 by feistus