vlachoudis/bCNC #296: Unwanted move after file open

Effer, 2016-04-12:
Not sure if always been present, but I've just noticed that if you open a file and soon after click the canvas the gcode is moved away.

vlachoudis, 2016-04-12:
Maybe I didn't quite understand what you mean, but I didn't notice any movement after loading the gcode.

Effer, 2016-04-12:
Sorry sometimes my grammar scares also me  ;)

Strange enough it appears **only using the shortcut**..
How to reproduce:
- open a file using the Ctrl+O shortcut
- the selected canvas tools is an undefined state (see picture)
- click somewhere on the canvas
- all gcode get moved as if move command is applyed

Using the open button everything works correctly

[image]

vlachoudis, 2016-04-12:
Thanks. I've just correct it. I was using the default Tkinter bindings <Key-o> for the origin command in the Canvas and the <Control-Key-o> for open file. Suprisingly enough with Ctrl-o both handlers were invoked.
Now I modified the Canvas to use the generic <Key> handler and I do the parsing, which works better.

closed 2016-04-12 by vlachoudis