vlachoudis/bCNC #297: optimize order issue

beta-tester, 2016-04-12:
hi,
bCNC is a very very useful tool, that i use on my old Raspberry Pi very successful.
i am extremely  thankful for that tool.

my DIY milling machine is very loud an slow, so i am always triyng to minimize the duration of milling. bCNC has an option to change the order of paths. but that option simply reorders the path blocks, that often still contains unnecessary big movements, because it keeps the direction of the path itself.
in my opinion, very often it would give better results, then the optimization algorithm would swap the direction of a path, to minimize necessary movements. then by swapping the direction of paths, then the milling machine can move bidirectional instead of only in one direction of the original source file.

e.g. if i have a g-file where on a PCB are many equal horizontal parallel lines.
all have the same direction from left to right.
then bCNC keeps that direction of all lines, and the result is all the time unnecessarily long movement from the right end of the line to the left begin of the next line.
better would be to swap the direction of some lines to move directional, that would shorten the time of milling.

and in a special case of 2D milling only, where every path is on the same level, it would be give potentially way better results, if the optimization process would split/taking apart paths at each (X, T, L) crossing point/edge and would complete reassemble the paths inclusive taking directional swap in to account...

possibly by giving the user the change to give different movements different weights/costs.

is something like this already in the pipeline?

ok, maybe it is only me, who see that issue - because i do my PCB outline milling files by hand with inkscape and the extension GCodeTools...

but it would be a extremely nice when bCNC would do optimize in that depth.

(currently i use bCNC version: 0.92, last change: 27.Mar.2016)

vlachoudis, 2016-04-12:
To make the optimization from both sides can be very easily done, but I didn't do intentionally. Path direction maybe is unimportant for PCB, but in general is important when milling. Depending the direction you will get different results. I've seen it also in the PCB milling I was getting more rougher edges and burr from one side.
A possible option would be to provide 2 optimization methods, one respecting the path direction and the other being free.
Now making a full optimization detecting crossings etc will be extremely time-costly for python :)

beta-tester, 2016-04-12:
to me, it would be already helpful, to have a button to swap a direction of a selected path. if you can also add a second optimization with bidirectional handling, that would even more helpful to me.
can you see the shine in my eyes...? :)

vlachoudis, 2016-04-12:
The button to swap the direction of a path is already there. 
In the Editor ribbon, look the ribbon group named "Route" on the right side, and you have three options there, Conventional, Climb and Reverse. These buttons alter the direction of the "cut".
Also you can visualize the path by clicking the "Info" button or the keyboard shortcut "n"

beta-tester, 2016-04-12:
oops.
and i always was pressing the button "invert" of the "order" group...
and nothing happens with the single selected path... now i know why.

thank you... maybe i can "optimize" my routing by hand.

is there still a chance to implement an option to automatically optimize paths/routes... :/

vlachoudis, 2016-04-13:
"Invert" is to invert the order of the g-code blocks are cut, not to "reverse" the direction of the cut.

beta-tester, 2016-04-13:
i used now the "Reverse" button for a while to kind of handmade optimization.
and i saw, that it will wipe out change Z an F in an unwanted manner.
e.g. from
path6773
```
(Start cutting path id: path6773)
(Change tool to Default tool)

G00 Z5.000000
G0 X-44.45 Y96.519992
G01 Z-0.125000 F65.0(Penetrate)
G1 X-44.45 Y89.534992 Z-0.125 F75
G1 X-34.29 Y89.534992 Z-0.125
G1 X-34.29 Y86.994992 Z-0.125
G1 X-26.67 Y86.994992 Z-0.125
G1 X-26.67 Y84.454992 Z-0.125
G1 X-24.13 Y84.454992 Z-0.125
G1 X-24.13 Y68.579992 Z-0.125
G1 X-24.13 Y66.039992 Z-0.125
G00 Z5.000000
(End cutting path id: path6773)
```
to
path6773 [reverse]
```
g0 x-24.13 y66.039992
g1 z0 f25
g1 x-24.13 y68.579992 f150
g1 x-24.13 y84.454992
g1 x-26.67 y84.454992
g1 x-26.67 y86.994992
g1 x-34.29 y86.994992
g1 x-34.29 y89.534992
g1 x-44.45 y89.534992
g1 x-44.45 y96.519992
g0 z3
```
you can see Z is changed from Z5 to Z3 and from Z-0.125 to Z0
and F is changed from F65 to F25 and from F75 to F150

shall i open a new issue for that? (if it is a bug)

vlachoudis, 2016-04-13:
I forgot to mention that. All operations that alter the path, change also the cutting properties and set the default ones used for the selected Stock and Material.
Also all other non-motion commands will be dropped.

beta-tester, 2016-04-13:
it will be dangerous to change the Z axis value. that can cause a head crash with the tool.
the reverse button is more or less useless, if i have to edit each path by hand.

PS.: i realize, why i never saw a difference on quality at if path is milling forth or backwards in the direction. i always mill only a single path, there it doesn't matter if the toll mills from left to right or from right to left. ;)

BTW.: importing inkscape files, how is that working? i don't get it work. it only loads the raw xml content to the editor in a useless format, it does not convert the inkscape content to any useful g-code.
i thought i could try do import a inkscape file directly to bCNC... and the open file file extension and the inkscape Editor button suggests to be able to do that... but i have no luck with that. the wiki doesn't describes the inkscape handling.

vlachoudis, 2016-04-13:
@beta-tester it is on the wiki maybe not on the best position. It is on the master page of the wiki.
You would need to convert the .svg from inkscape to .eps
```inkscape -E file.eps file.svg```
Then to convert to dxf with pstoedit
```pstoedit -f 'dxf: -mm' file.eps file.dxf```
There are some options to control the accuracy scaling etc...

beta-tester, 2016-04-13:
ok... i think i have to play whit my workflow to get my way with bCNC...
i keep quiet for the moment...
thank you for taking the time to help me.

closed 2016-04-13 by beta-tester