vlachoudis/bCNC #298: Black Box in Maschine Settings

BlueOrangeLive, 2016-04-13:
Hallo 
ich habe heute ein Update gemacht (13.04.2016)
Dabei ist mir beim Anklicken meiner Maschinen Einstellungen in den beiden letzten Zeilen: Header G-Code und Footer G-Code aufgefallen das ich nicht normal Editieren kann. Eine schwarze Box Erscheint. Wenn ich Tasten Drücke kommen keine Zeichen aber nach dem drücken von Enter Werden sie Eingefügt.

Gruß aus Kevelaer Jürgen

bCNC ist ein Tolles Programm.
---------------------------------------------------------------

Hello 
I made an update today (04/13/2016)
Header G code and Footer G code noticed that I can not edit normal: That reminds me settings in the last two lines is click on my machine. A black box appears. Keystrokes If I get no mark but after pressing Enter Will they Insertion.

Greetings from Kevelaer Jürgen

bCNC is a Great program.
[image]

vlachoudis, 2016-04-13:
Can you provide some more information, what OS you have, python version etc...

vlachoudis, 2016-04-13:
Of course in the mean time you can change the values in the ${HOME}/.bCNC file

BlueOrangeLive, 2016-04-13:
ich nutze fedora23-linux mit Python 2.7.11
In der letzten Version von bCNC funktionierte es noch. Es sind ja auch nur die beiden letzten Eingabefelder.

I use fedora 23-linux with Python 2.7.11
In the latest version of bCNC it still worked. There are indeed only the last two fields.

vlachoudis, 2016-04-14:
I was able to reproduce it. Now it is fixed. You have to download the new master

BlueOrangeLive, 2016-04-14:
Thank you, it works.

Next topic: millimeter
I use only mm, not inches. In machines not cross setting in inches.
OK Start bCNC, Connect with grbl, pressures Stop button appears at Controll State Units - INCHES instead mm. The machine performs even everything out in INCHES.
Exit bCNC.
OK Start bCNC New, Connect with grbl, pressures HOME button, then stop button, then Units in mm. Everything is OK.
Allso only if first stop button is pressed, the error occurs.
I can live with it, to be careful just stop.

FYI: I use an Arduino Mega2560 with private Shield.
  Software of https://github.com/gnea/grbl-Mega/
or my in-paste version https://github.com/BlueOrangeLive/grbl-jog-mega2560
  of https://github.com/heise/GRBL-JOG   = Grbl 0.9j/jog

Thank you

Danke, es klappt. 

Nächstes Thema: Millimeter
Ich nutzt nur mm, keine Inches. In Maschinen setting kein Kreuz bei Inches.
OK, Starte bCNC , Verbinde mit  grbl ,  Drücke Stop Taste, dann erscheint bei Controll State Units - INCHES anstatt mm . Die Maschine führt auch dann alles in INCHES aus.
Verlasse bCNC.
OK, Starte bCNC Neu , Verbinde mit grbl , Drücke HOME Taste , dann Stop Taste  ,dann ist Units in mm. Alles ist OK. 
Allso nur wenn zuerst Stop Taste gedrückt wird, kommt der Fehler.
Ich kann damit leben, muß eben halt aufpassen.

Zur Info: Ich benutze ein Arduino Mega2560 mit eigenem Shield. 
 Software von https://github.com/gnea/grbl-Mega/  
oder meine angepaste Version  https://github.com/BlueOrangeLive/grbl-jog-mega2560
 von https://github.com/heise/GRBL-JOG

vlachoudis, 2016-04-14:
@BlueOrangeLive This one I cannot reproduce. @Effer also mentioned that. Could you send me a screenshot or a copy of the "Terminal" maybe I can figure out from there.

Effer, 2016-04-14:
Here my console:

ok
ok
ok
ok
G90
G0X0Y0S12000M3
G0X-31.2974Y-10.5512Z5
G1Z-1F12000
G1X-31.8728Y-10.1741F72000
G1X-31.2992Y-10.5578
G1X-31.1073Y-10.6676
G1X-31.2974Y-10.5512
ok
ok
G1X-30.6189Y-9.4805
ok
G1X-31.8977Y-8.6423
ok
ok
G1X-33.2921Y-7.679
ok
G1X-34.8044Y-6.5896
ok
G1X-36.436Y-5.3735
G1X-38.1881Y-4.0301
ok
G1X-40.0612Y-2.5591
ok
G1X-42.0505Y-0.9645
ok
G2X-41.6747Y1.7369I0.9929J1.2387
ok
ok
G1X-40.509Y2.2286
ok
G1X-39.4388Y2.7373
ok
G1X-38.4244Y3.2798
ok
G1X-37.4919Y3.841
ok
G1X-36.6148Y4.4352
G1X-35.8006Y5.0568
ok
Grbl 0.9i ['$' for help]
G20 G54
$G
G43.1Z0.0
$G
ok
[G0 G54 G17 G20 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
[G0 G54 G17 G20 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
ok
[G0 G54 G17 G20 G90 G94 M0 M5 M9 T0 F0. S0.]
ok

vlachoudis, 2016-04-14:
@Effer could you send also a $G command before the stop and check the output

Effer, 2016-04-14:
Well, I've notice that when I start bCNC state is undefined:

[image]

It become populated only if I force a $G from terminal. After that the switch to inch never happen again.
For some reason $G is not polled correctly.

vlachoudis, 2016-04-14:
I had for some reason as initial values of $G the G20 G54, so if CNC.vars["G"] was not populated (it is done in regular intervals if there is activity in the serial line) it was restoring the wrong init settings.
Now it is corrected.

Effer, 2016-04-14:
Now seems fixed.
Thanks to @BlueOrangeLive to point out that it happens only on first time.
And obviusly to @vlachoudis to fix the problem!

closed 2016-04-14 by vlachoudis

BlueOrangeLive, 2016-04-14:
You're fast . Thank you, it works.

vlachoudis, 2016-04-14:
@BlueOrangeLive thank you for reporting it :)