vlachoudis/bCNC #299: Drawing timeout implementation and Windows compatibility

kkugelblitz, 2016-04-14:
As it relies on SIGALARM, the drawing timeout implementation 
https://github.com/vlachoudis/bCNC/commit/93151f6fc3c1e4acab41ccd71977919611df3971
seems to break compatibility with MS Windows. 

Personally I am ok with Linux, however it would be nice to keep all options open (and I am sure there are windows-only users of this great piece of SW around here).
There is a stackoverflow article on the topic, but obviously the proposed solution is less elegant than signaling:
http://stackoverflow.com/questions/8420422/python-windows-equivalent-of-sigalrm

Effer, 2016-04-14:
Confirmed @kkugelblitz , it broke Windows compatibility.

vlachoudis, 2016-04-14:
@Effer you are using windows, could you check if the proposed solution works?

Effer, 2016-04-14:
Well @vlachoudis  the stackoverflow code run as expected.
Not sure what the best way to integrate in the existing code. If you need more test let me know.

vlachoudis, 2016-04-14:
I don't like adding a separate thread just for timekeeping. I can periodically check the timeout inside the heavy loop

vlachoudis, 2016-04-14:
I've replaced the signal handler with timekeeping using the simple time.time() command. I've made some benchmarking and it doesn't add any noticeable penalty on the rendering.
Could you please check on windows

Effer, 2016-04-14:
Fixed also this.
Nice the combo box with the choice for timeout. Do you believe in the future will be possible to suspend the drawing, and then resume it in background?

vlachoudis, 2016-04-14:
The problem is that Tk is not compiled with be re-entrant in some linux distributions like fedora. So it can only work on the master thread.

So if it works ok I close the issue

closed 2016-04-14 by vlachoudis