vlachoudis/bCNC #300: If GRBL is reporting alarm bCNC doesn't connect

burtbick, 2016-04-16:
bCNC is great and is a big help with the Mostly Printed CNC machine I'm building but I have an issue that is annoying.

When my controller powers up most of the time I get an alarm report (probably due to caps on the limit switches charging).  When this happens bCNC says not connected when trying to open the port.  

I work around this by closing the port in bCNC and then opening it with minicom.  In minicom on connection I get the GRBL prompt (and the message that it is locked due to alarm).  So then I issue $X and close minicom.  Then connect in bCNC and now bCNC can connect.

Would it be possible for bCNC to recognize that GRBL is in alarm (but can actually connect)?  This would allow me to then issue the $X command from bCNC rather than the extra steps.

I didn't see this mentioned but may have missed it in the Issues list.

Regards,
Burt

vlachoudis, 2016-04-17:
Hi @burtbick. The two things are different. bCNC reports connected when the serial port is opened successfully independent of the state of grbl. Then if grbl reports alarm, bCNC will display alarm, however you should be able to send the $X or $H or reset since you are connected.

burtbick, 2016-04-17:
Thanks, that is what I would have expected.  But that doesn't appear to be what is happening.  When I power up the controller and it comes up with GRBL reporting an alarm bCNC does open the serial port but the status is still showing Not connected on an Orange background above WPos.  And it is not appearing to send the $X command.

But if I Close the port in bCNC and open it in Minicom I get the GRBL prompt and can enter $X in Minicom, then close Minicom and click Open back in bCNC and then Status changes to Connected.

I have to be gone for most of the day today, but I'll try to gather more data to see what specifically is going on.  Maybe I'll add some debug code to trace what is actually happening in bCNC in this case.

Burt

vlachoudis, 2016-06-09:
Any news on this issue?

burtbick, 2016-06-13:
I have not seen the problem recently, but I haven't had time to do much testing with the CNC machine since I was rebuilding the Gantry.I will do some more testing over the next week or two and see if it happens again.Burt

connect (#300)
<mention@noreply.github.com>

vlachoudis, 2016-06-25:
OK I am closing the issue and if it reappears please open it again

closed 2016-06-25 by vlachoudis

burtbick, 2016-06-25:
Will do.  Not sure why it went away, but good thing that it did:)

quornian, 2016-09-29:
For anyone else having similar intermittent issues when trying to establish a connection, it turned out for me to be an old version of pySerial on the system (from 2010). Putting a new one in the lib/ directory of bCNC fixed this.

That is, I cloned https://github.com/pyserial/pyserial, took the "serial" folder out and moved it to bcnc/lib/serial. Now I get a connection every time. If I remove it, the issue resurfaces.

A little debugging showed that I was getting 4 bytes of junk from the serial connection when first established, which threw everything else off. Now I correctly get: "Grbl 0.9g ['$' for help]".

vlachoudis, 2016-09-29:
@quornian which version is creating the problem?

quornian, 2016-09-30:
On Linux Mint 17.3, I have:
```
$ dpkg -s python-serial
Package: python-serial
Status: install ok installed
Priority: optional
Section: python
Installed-Size: 403
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Source: pyserial
Version: 2.6-1build1
Depends: python (>= 2.7), python (<< 2.8), python:any (>= 2.7.1-0ubuntu2)
Suggests: python-wxgtk2.8 | python-wxgtk
Description: pyserial - module encapsulating access for the serial port
 This module encapsulates the access for the serial port. It provides
 back-ends for standard Python running on Windows, Linux, BSD (possibly
 any POSIX compliant system). The module named "serial" automatically
 selects the appropriate back-end.
Original-Maintainer: Matthias Klose <doko@debian.org>
Homepage: http://pyserial.sourceforge.net/
Python-Version: 2.7
```
```
$ head -n7 /usr/lib/python2.7/dist-packages/pyserial-2.6.egg-info
Metadata-Version: 1.1
Name: pyserial
Version: 2.6
Summary: Python Serial Port Extension
Home-page: http://pyserial.sourceforge.net/
Author: Chris Liechti
Author-email: cliechti@gmx.net
```

onekk, 2016-09-30:
I have solved my problem installing the pip version of pyserial that upgrade to 3.something