vlachoudis/bCNC #301: {Feature Request}

atrueresistance, 2016-04-17:
Is it possible when the Z axis limit switch is hit that power not be disconnected to the stepper motors? (I see where this would be useful in probing -z)

This is an example. Gcode generated in Fusion 360 .2" clearence. As you can see I'm very close to the working limit of the SO3 machine Z wise. Since I had something clamped on the left side of my machine, I strapped this piece down more towards the right. Once the run was done, it moves back to home, but hit the limit switch sending the spindle downward because the SO3 Z weight. 
[image]

CarlosGS, 2016-04-17:
This seems like an issue with the gcode that you are using
Please check the last lines of the g-code file for that piece, and see if there is a M0 or M18 command

Just in case, take a look as well to the "gcode footer" option in bCNC->Machine settings (though by default it does not turn off stepper motors)

CarlosGS, 2016-04-17:
Also, the part looks very nice! What kind of wood is it?

atrueresistance, 2016-04-17:
X-0.865 Y13.284 Z-30.531 I-0.007 J0.635
    G0 Z5.08
    M9
    G28 G91 Z0
    G28 X0 Y0
    M30

This is the end of the file sent that caused the Z limit switch collision. 

I'll take a look at the gcode footer in bCNC, but don't recall ever editing that before.

atrueresistance, 2016-04-17:
I think it was the bCNC file. If I click on Tools -> Machine I can see the Footer gcode is `M5G0 Z20`. When I click on it I can't change it though.

vlachoudis, 2016-04-21:
@atrueresistance please update the bCNC there was a bug and it was not permitting to change it.

atrueresistance, 2016-04-30:
@vlachoudis  After updating I was able to change it. I feel as though when you click it to change it should change to a textbox with cursor and allowing you to see the text right away as you are typing.

closed 2016-04-30 by atrueresistance