vlachoudis/bCNC #303: G-Code Errors

Guenni75, 2016-04-19:
Im using bCNC a few days and since yesterday i have problems with G-Code like this:

G21 G90 G64 G40
G0 Z3.0
( T3 : 3.0 )
T3 M6
( Profil1 )
G17
M3 S1000
G0 X0.0 Y-1.5
G0 Z1
G1 F300.0 Z-0.4
G1 F200.0 X8.5
G2 X11.0 Y-4.0 I0.0 J-2.5
.........

The last G2 causes an error from the GRBL Controller.
I tryed the same Code with other G-Code senders and it works fine.
So it could not be an error in the Code!
The error only appears if i am sending with bCNC.

Any Ideas?

vlachoudis, 2016-04-20:
Are you sure that is the G2 causing the problem and not the T3 M6?
What error message is displayed. You can always click on the error message and an extensive explanation will be shown.

Most of the senders silently ignore the errors from grbl and continue which can be dangerous.
If it is the M6 tool change command. bCNC has various ways to treat it
Go to the "Probe -> Tool" and change the "Policy" of tool change
you have several options there. If you want to have similar behavior like the other senders
select the "Ignore M6 commands", or you can setup your machine for a manual tool change and calibration. Look on the bCNC wiki how it works.

Guenni75, 2016-04-20:
Bcnc ignores errors too!!!
Luckily i trie the codes without a tool and often bcnc drives crazy around with errors!
The same code works with G-Code sender and the result is perfect.

I think it was the G2, because at this point the router stops (luckily) at x 8.5 y -1.5.

I try it again tomorrow and post the error and terminal output.

vlachoudis, 2016-04-21:
The G2 can only cause an error if there is a precision problem in the coordinates and radius. bCNC is anyhow sending all commands as is to grbl and checks if grbl reports an error. It doesn't do any gcode checking.

Guenni75, 2016-04-22:
Ok, but the router didn't stop immediately and that is dangerous.

G64 is the Problem!!!
The same G-Code without G64 in the header and it works!!

CamBam Users, watch out for G64!!

vlachoudis, 2016-04-22:
@Guenni75 thanks for finding out

closed 2016-04-22 by vlachoudis