vlachoudis/bCNC #307: Fit to screen

kworsfold, 2016-04-22:
Hi, 
Great project 

I have loaded on the RPi and works flawlessly, however I a have a minor problem with the Screen size, currently 5 inch 800x480, currently in the process of upgrading to 7 inch 1024x800.
I have managed to Autoload on boot of RPi, but does not default to Full Screen Mode, and also unable to UnDecorate to maximize view, as part of the screen is hidden at the bottom.
Do you know how I can set it up to load into Full Screen Mode (Maximize) and UnDecorate.
I have set up ".config/openbox/lxde-pi-rc.xml" to do this but does not work, possibly to do with the file name/Python Script.

Any suggestions, and what is the correct file name to change parameters in ".config/openbox/lxde-pi-rc.xml"

vlachoudis, 2016-04-22:
To undecorate depends on your window manager.
In bCNC you can force the size with the "-g" command line option
e.g.   bCNC -g 800x480
The last window dimensions are saved inside the ~/.bCNC file as "width" and "height", "x", "y"
you can always tweak those values to fit exactly your screen

closed 2016-05-03 by vlachoudis

Cncninja, 2020-12-19:
I keep getting unknow command when trying this