vlachoudis/bCNC #308: Suggested workflow doc for new PCB routing/drilling users.

ErikShorb, 2016-04-29:
First time using bCNC, took a bit of trial and error to understand required steps for success. I'd suggest adding something like below to wiki to help first time users.  Regards.
-----------------------------------------
1. Clean Copperclad PCB with steel wool or equivalent until shiny.  If you skip this step, you may damage a tool bit when probing.

2. Clamp Copperclad PCB to waste board - want PCB to be held down "flat" and secure.  Double-backed tape works well for some.

3.  Mount First Drill bit in Router 

4.  Roughly locate the Router/bit to starting location

5.  Start bCNC

6.  Connect to CNC via bCNC

   a. Select File Menu
   b. Select Serial Port
   c. Click on Open 
   d. Select Control Menu
   e. Click on Unlock  

7.  Prepaare for Autoleveling 

   a. Select File Menu
   b. Open gcode file for cutout routing - this will make it simple to setup the correct Autolevel area
   c. Position bit to X and Y start location using Control widgets.
   d. Select Probe Menu
   e. Click on Probe Submenu
   f. Zero out X and Y by Clicking on X=0 the Y=0 
   g. Click on "fit to screen" button
   h. Lower tool bit within a couple of mm of PCB using jog controls
   i. Zero out Z by Clicking on Z=0
   j. In Probe menu, Probe section, Set the Z Pos: to -10 - this will allow the bit to go 10mm down to probe before setting off an alarm.
   k.  ATTACH PROBE CLIPS TO PCB AND TOOL BIT   - FAILURE TO DO SO BEFORE PROCEEDING MAY RESULT IN A DAMAGED BIT.
   l. Click Probe button (in Probe: section)
   m. Zero out Z axis - Click on Z=0
   n. Raise Z by 1mm using jog

8.	Autolevel
	
	a. Under Probe Menu, Select Autolevel Submenu
	b. Click on Margins - This will configure an autolevel area mapped to you gcode file limits
	c. modify number of X/Y test points as needed by adjusting  N for X and Y
	d. set Z Min to -10 and Z Max to 0.5 - if you expect a large deviation in height across the board, set Z max to a higher value ( and perhaps consider leveling waste board and/or machine)
	e.  CONFIRM PROBE CLIPS TO PCB AND TOOL BIT   - FAILURE TO DO SO BEFORE PROCEEDING MAY RESULT IN A DAMAGED BIT.
	f. Click on Scan
	g. Optional - Return to X/Y zero  - in Command: (at bottom) enter G0 X0 Y0
	h. REMOVE PROBE CONNECTION TO TOOL BIT.

9. Load first Drill gcode file

	a. Under File menu select Open and select drill file
	b. Answer no to the "Probe File modified" dialog
	c. Answer no to the "Existing Autolevel" dialog
	d. Optional - Select ISO view from "Change viewing Angle" widget.

10. Perform Drilling operation

	a. Confirm that you've REMOVED THE PROBE CONNECTION TO THE TOOL BIT.
	b. If manually controlling power to your router, turn it on now.
	c. Select Control menu
	d. Click on Start.
	e. When drilling is complete, if manually controlling power to your router, turn it off.

11. Change Tool Bit - Do this every time you change a bit, whether it be a drill, routeing, or engraving bit.

	a.  Raise Z to comfortable height using jog controls.
	b.  Replace Tool bit
	c.  ATTACH PROBE CLIPS TO PCB AND TOOL BIT
	d.  Lower bit to within a couple of mm of PCB using jog controls.
	e.  Select Probe Menu
	f.  Select Probe Submenu
	g.  Zero out Z by Clicking Z=0
	h.  Click on Probe button
	i.  Zero out Z by Clicking Z=0
	j.  Raise Z by 1mm using jog controls.
	k.  REMOVE PROBE CONNECTION TO TOOL BIT.

12. If performing more Drilling operations, follow steps 10-12, with final Tool bit change being to the trace routing bit.

13. Load trace routing gcode file.

	a. Under FIle menu select Open and select trace routing gcode file
	b. Answer no to the "Probe File modified" dialog
	c. Answer no to the "Existing Autolevel" dialog
	d. Optional Select preferred viewing angle.

14. Change Tool Bit per step 11.

15. Perform Trace Routing operation

	a. Confirm that you've REMOVED THE PROBE CONNECTION TO THE TOOL BIT.
	b. If manually controlling power to your router, turn it on now.
	c. Select Control menu
	d. Click on Start.
	e. When routing is complete, if manually controlling power to your router, turn it off.

16. Load Outline routing gcode file.

	a. Under FIle menu select Open and select outline routing gcode file
	b. Answer no to the "Probe File modified" dialog
	c. Answer no to the "Existing Autolevel" dialog
	d. Optional - Select preferred viewing angle.

17. Change Tool Bit per step 12.

18. Perform outline Routing operation

	a. Confirm that you've REMOVED THE PROBE CONNECTION TO THE TOOL BIT.
	b. If manually controlling power to your router, turn it on now.
	c. Select Control menu
	d. Click on Start.
	e. When routing is complete, if manually controlling power to your router, turn it off.

--------------------------------

vlachoudis, 2016-05-02:
Thanks for the detailed step-by-step guide. I've created under tutorials a like to
https://github.com/vlachoudis/bCNC/wiki/PCBMillingStepByStep
Could you please insert it there and format it.

Effer, 2016-05-02:
@vlachoudis I saw that @flaplante has already kindly published this page:
https://github.com/vlachoudis/bCNC/wiki/PCB-101---First-time

(and he also correct a lot of my bad grammar mistakes, many thanks :)

Ozzy62, 2016-05-03:
Thank you, so much for creating this. This has been missing for so long.

vlachoudis, 2016-05-03:
Thanks I close the issue

closed 2016-05-03 by vlachoudis